MarlinFirmware/Marlin #19651: Creality 4.2.7 Board in Ender 3 Mainboad fan not running

GdLakeIoT, 2020-10-07:
Hi,
I just upgraded to the new 4.2.7 board and used the bug fix marling firmware. Everythings seems to work, but the Mainboard fan is not spinning, regardless of what I print. The fan is in the K-FAN2 port.
I can't find the code in the .h files that sets this Fan

ellensp, 2020-10-07:
The only sofware controled fan Marlin knows of is fan (next to power in) on PA0
I dont see any aditional mosfet..  I stronly suspect these K-FAN1 and K-FAN2 are just power plug for always on fans

thisiskeithb, 2020-10-07:
Check your wiring/plug in your fans again. Since these aren't controllable by firmware, there is something wrong likely with the board/wiring.

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-07 by thisiskeithb

mseddon, 2020-11-21:
Fwiw, I installed the 4.2.7 today and built marlin for it. I can confirm that K-FAN1 and K-FAN2 are BOTH connected to the PWM, which probably caused the confusion- the mainboard fan does not turn on until the hotend fan kicks in on this board.

leviustinov, 2020-11-22:
> Fwiw, I installed the 4.2.7 today and built marlin for it. I can confirm that K-FAN1 and K-FAN2 are BOTH connected to the same PWM (which Marlin correctly uses), which probably caused the confusion- the mainboard fan does not turn on until the hotend fan kicks in on this board.

Did you ever find out how to get to be always on? Not all materials need cooling.

LordMM74, 2021-01-03:
I had same problem.
Does exists any solution?

GdLakeIoT, 2021-01-03:
You could connect it to the 24v terminal that supplies power to the main board?

leviustinov, 2021-01-03:
> You could connect it to the 24v terminal that supplies power to the main board? Get Outlook for Android<https://aka.ms/ghei36>
> […](#)

I thought of that, but then what amperage is that line rated for?...

GdLakeIoT, 2021-01-03:
I am guessing that the amps of the fan is marginal
I dont have access to my printer or extra fans to check

________________________________ From: LordMM74 notifications@github.com<mailto:notifications@github.com> Sent: Sunday, January 3, 2021 10:55:00
AM To: MarlinFirmware/Marlin Marlin@noreply.github.com<mailto:Marlin@noreply.github.com> Cc: George Lake georgedlake@outlook.com<mailto:

samtny, 2021-01-15:
Huh. I have the same board and I could have sworn the mainboard fan was turning on at times when the pwm fan was off. Now I am working with another firmware and it is clear the mainboard fan stays off until the pwm / hotend fan is turned on.

May look into connecting it to 24v terminal then as I am printing PETG without a fan with good results.