MarlinFirmware/Marlin #19653: Ender 3 V2 DWIN Display MarlinUI Building Fails

lgndns, 2020-10-07:
Hi :) I'm having so much errors when i try to use the MarlinUI for my Ender 3 V2 Controller.

Error From the Terminal:
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\dwin\e3v2\rotary_encoder.cpp.o
In file included from .pio\libdeps\STM32F103RET6_creality\LiquidCrystal/LiquidCrystal.h:45:0,
                 from Marlin\src\lcd\HD44780\ultralcd_HD44780.h:96,
                 from Marlin\src\lcd\HD44780\lcdprint_hd44780.cpp:22:
.pio\libdeps\STM32F103RET6_creality\LiquidCrystal/FastIO.h:69:0: warning: "ATOMIC_BLOCK" redefined
 #define ATOMIC_BLOCK(dummy) if(true)

In file included from Marlin\src\lcd\HD44780\../../inc/../HAL/./STM32F1/HAL.h:41:0,
                 from Marlin\src\lcd\HD44780\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\lcd\HD44780\../../inc/MarlinConfig.h:30,
                 from Marlin\src\lcd\HD44780\../ultralcd.h:24,
                 from Marlin\src\lcd\HD44780\lcdprint_hd44780.cpp:19:
C:\Users\nuret\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple\include/util/atomic.h:51:0: note: this is the location 
of the previous definition
 #define ATOMIC_BLOCK(type) \
 
In file included from .pio\libdeps\STM32F103RET6_creality\LiquidCrystal/LiquidCrystal.h:45:0,
                 from Marlin\src\lcd\HD44780\ultralcd_HD44780.h:96,
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\extui\lib\anycubic_chiron\FileNavigator.cpp.o
                 from Marlin\src\lcd\HD44780\lcdprint_hd44780.cpp:22:
.pio\libdeps\STM32F103RET6_creality\LiquidCrystal/FastIO.h:70:0: warning: "ATOMIC_RESTORESTATE" redefined
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\extui\lib\anycubic_chiron\Tunes.cpp.o
 #define ATOMIC_RESTORESTATE
 
In file included from Marlin\src\lcd\HD44780\../../inc/../HAL/./STM32F1/HAL.h:41:0,
                 from Marlin\src\lcd\HD44780\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\lcd\HD44780\../../inc/MarlinConfig.h:30,
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\extui\lib\anycubic_chiron\chiron_tft.cpp.o
                 from Marlin\src\lcd\HD44780\../ultralcd.h:24,
                 from Marlin\src\lcd\HD44780\lcdprint_hd44780.cpp:19:
C:\Users\nuret\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple\include/util/atomic.h:54:0: note: this is the location 
of the previous definition
 #define ATOMIC_RESTORESTATE \
 
In file included from .pio\libdeps\STM32F103RET6_creality\LiquidCrystal/LiquidCrystal.h:45:0,
                 from Marlin\src\lcd\HD44780\ultralcd_HD44780.h:96,
                 from Marlin\src\lcd\HD44780\ultralcd_HD44780.cpp:34:
.pio\libdeps\STM32F103RET6_creality\LiquidCrystal/FastIO.h:69:0: warning: "ATOMIC_BLOCK" redefined
 #define ATOMIC_BLOCK(dummy) if(true)

In file included from Marlin\src\lcd\HD44780\../../inc/../HAL/./STM32F1/HAL.h:41:0,
                 from Marlin\src\lcd\HD44780\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\lcd\HD44780\../../inc/MarlinConfig.h:30,
                 from Marlin\src\lcd\HD44780\ultralcd_HD44780.h:28,
                 from Marlin\src\lcd\HD44780\ultralcd_HD44780.cpp:34:
C:\Users\nuret\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple\include/util/atomic.h:51:0: note: this is the location 
of the previous definition
 #define ATOMIC_BLOCK(type) \
 
In file included from .pio\libdeps\STM32F103RET6_creality\LiquidCrystal/LiquidCrystal.h:45:0,
                 from Marlin\src\lcd\HD44780\ultralcd_HD44780.h:96,
                 from Marlin\src\lcd\HD44780\ultralcd_HD44780.cpp:34:
.pio\libdeps\STM32F103RET6_creality\LiquidCrystal/FastIO.h:70:0: warning: "ATOMIC_RESTORESTATE" redefined
 #define ATOMIC_RESTORESTATE
 
In file included from Marlin\src\lcd\HD44780\../../inc/../HAL/./STM32F1/HAL.h:41:0,
                 from Marlin\src\lcd\HD44780\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\lcd\HD44780\../../inc/MarlinConfig.h:30,
                 from Marlin\src\lcd\HD44780\ultralcd_HD44780.h:28,
                 from Marlin\src\lcd\HD44780\ultralcd_HD44780.cpp:34:
C:\Users\nuret\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple\include/util/atomic.h:54:0: note: this is the location 
of the previous definition
 #define ATOMIC_RESTORESTATE \
 
Marlin\src\lcd\HD44780\ultralcd_HD44780.cpp:99:17: error: 'LCD_PINS_RS' was not declared in this scope
   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7);
                 ^~~~~~~~~~~
Marlin\src\lcd\HD44780\ultralcd_HD44780.cpp:99:17: note: suggested alternative: 'LCD_PINS_D5'
   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7);
                 ^~~~~~~~~~~
                 LCD_PINS_D5
                              ^~~~~~~~~~~~~~~
Marlin\src\lcd\HD44780\ultralcd_HD44780.cpp:99:30: note: suggested alternative: 'LCD_PINS_D5'
   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7);
                              ^~~~~~~~~~~~~~~
                              LCD_PINS_D5
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\HD44780\ultralcd_HD44780.cpp.o] Error 1
========================================================== [FAILED] Took 46.84 seconds ========================================================== 

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:46.840
===================================================== 1 failed, 0 succeeded in 00:00:46.840 =====================================================

ETE-Design, 2020-10-07:
That's because it isen't supported yet... Look here: #19371

thisiskeithb, 2020-10-07:
Closing since this is an alpha feature. Subscribe to https://github.com/MarlinFirmware/Marlin/pull/19371 for updates on progress since it's still in Alpha support.

closed 2020-10-07 by thisiskeithb