MarlinFirmware/Marlin #19658: SKR V1.4

CharlyAxams, 2020-10-07:
//file Configuration.h Release:2.0.7
//section info
#define STRING_CONFIG_H_AUTHOR "(Charly)"
//section machine
#define SERIAL_PORT_2 -1
#define BAUDRATE 115200
#define BLUETOOTH 
#define MOTHERBOARD BOARD_BTT_SKR_V1_4
#define AUTO_POWER_CONTROL 
#define COREXY 
#define INVERT_Y_DIR false
#define INVERT_Z_DIR true
#define X_BED_SIZE 240
#define Y_BED_SIZE 240
#define Z_MAX_POS 175
#define FILAMENT_RUNOUT_SENSOR 
#define FILAMENT_RUNOUT_DISTANCE_MM 25
//section extruder
//#define HOTEND_OFFSET_X { 0.0, 0.00 }
//#define HOTEND_OFFSET_Y { 0.0, 0.00 }
//section temperature
#define TEMP_SENSOR_BED 1
#define PID_EDIT_MENU 
#define PID_AUTOTUNE_MENU 
#define PID_PARAMS_PER_HOTEND 
//#define DEFAULT_Kp 63.0
//#define DEFAULT_Ki 2.25
//#define DEFAULT_Kd 440
#define PIDTEMPBED 
//#define DEFAULT_bedKp 100.64
//#define DEFAULT_bedKi 14.82
//#define DEFAULT_bedKd 455.69
#define PREHEAT_1_TEMP_HOTEND 210
#define PREHEAT_1_TEMP_BED 50
#define NOZZLE_PARK_FEATURE 
#define PRINTCOUNTER 
//section homing
//#define ENDSTOPPULLUPS 
#define Z_MIN_ENDSTOP_INVERTING true
#define Z_MIN_PROBE_ENDSTOP_INVERTING true
#define X_DRIVER_TYPE TMC2208
#define Y_DRIVER_TYPE TMC2208
#define Z_DRIVER_TYPE DRV8825
#define E0_DRIVER_TYPE DRV8825
#define Z_SAFE_HOMING 
#define Z_SAFE_HOMING_X_POINT 4
#define Z_SAFE_HOMING_Y_POINT 4
//section motion
#define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 800, 277.77 }
#define DEFAULT_MAX_ACCELERATION { 1500, 1500, 100, 10000 }
#define DEFAULT_ACCELERATION 1500
#define DEFAULT_TRAVEL_ACCELERATION 1500
#define CLASSIC_JERK 
//section probes
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN 
#define FIX_MOUNTED_PROBE 
#define NOZZLE_TO_PROBE_OFFSET { 0, 30, 0 }
#define PROBING_MARGIN 0
#define XY_PROBE_SPEED 1500
#define PAUSE_BEFORE_DEPLOY_STOW 
//section calibrate
#define AUTO_BED_LEVELING_BILINEAR 
#define RESTORE_LEVELING_AFTER_G28 
#define GRID_MAX_POINTS_X 4
#define GRID_MAX_POINTS_X 4
#define GRID_MAX_POINTS_X 4
//section extras
#define EEPROM_SETTINGS 
#define FAN_SOFT_PWM 
//#define PRINTER_EVENT_LEDS 
//section lcd
#define LCD_LANGUAGE de
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 

//file Configuration_adv.h Release:2.0.7
//section temperature
#define WATCH_BED_TEMP_PERIOD 360
//section extras
//#define ARC_SUPPORT 
#define ADVANCED_PAUSE_FEATURE 
#define MEASUREMENT_DELAY_CM 4
#define MMU2_SERIAL internalSerial
//section lcd
#define BABYSTEPPING 
//section leveling
#define PROBING_MARGIN_LEFT PROBING_MARGIN
#define PROBING_MARGIN_RIGHT PROBING_MARGIN
#define PROBING_MARGIN_FRONT PROBING_MARGIN
#define PROBING_MARGIN_BACK PROBING_MARGIN

thisiskeithb, 2020-10-07:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-07 by thisiskeithb

sjasonsmith, 2020-10-07:
@CharlyAxams why do you continue opening issues without valid descriptions? Please stop!