MarlinFirmware/Marlin #19659: [FR] Support for filament motion sensors

logiclrd, 2020-10-07:
### Description

From what I can see, there has been discussion on and off for a long time now about sensors that detect, accurately and in real time, actual motion of filament, so that these could be used to provide not only filament runout detection but also detection of blockages, stripping of the filament and so on. There is now at least one readily-available component on the market that does this in a mass-produced way:

https://www.duet3d.com/MagneticFilamentMonitor

I also am intrigued by the idea of using cheap optical mouse sensors, and have half a plan to try to design something that uses multiple sensors to make up for the fact that they don't have 100% reliability.

The information these sensors provide is proportional to the quantity of filament passing through the sensor. The sensor is measuring the rotation of a wheel being turned by the filament as it passes, and returns the change in angle of that wheel regularly. Duet3D's sensor returns the wheel's angle measured in the range `[0, 1024)` for a full rotation, and provides 25 samples per second. With a simple calculation (and the right calibration), these values would allow fairly straightforward testing that filament is actually moving in accordance with the commands sent to the extruder stepper.

(I speculate that the last mile would probably be the least straightforward, in that some buffer of expected filament moves would need to be maintained, and a trigger mechanism for a filament problem would need to fire only after too much time had elapsed without the measured movement matching the movement actually sent to the E stepper, since there'd be some lag between the E movement command being processed, the stepper motor actually moving, and the sensor returning a corresponding movement, not to mention noise in the signal.)

### Feature Workflow

1. Configure a printer with a filament motion sensor of some sort wired in. 
2. Enable processing of the data in the Marlin build, and calibrate the values appropriately.
3. Enable the feature at runtime.
4. If filament breaks or jams mid-print, then Marlin detects it because of the abnormal angle deltas and can run a recovery script.

#### Additional Information

Duet3D has extensive documentation of their particular monitor here: https://duet3d.dozuki.com/Wiki/Duet3dFilamentMonitor_RotatingMagnetVersion

In particular:

> Bits 0-9 represent the magnet angle measured in units of 1/1024 of a complete rotation.
> 
> The 16 bits are sent using a self-clocking protocol as follows. The nominal bit time is 1ms but the Duet firmware allows a reasonable tolerance because the filament monitor clock source is the internal R-C oscillator in the microcontroller, so its speed is somewhat variable.
> 
> * Idle state: the line must be be at 0 for at least 8 bit times
> * Start bits: 1 followed by 0
> * Data bits 15, 14, 13, 12
> * Stuffing bit (inverse of bit 12)
> * Data bits 11,10,9,8
> * Stuffing bit (inverse of bit 8)
> * Data bits 7,6,5,4
> * Stuffing bit (inverse of bit 4)
> * Data bits 3,2,1,0
> * Stuffing bit (inverse of bit 0)
> * After the last stuffing bit, the line returns to 0 until the next start bit.

If this feature assumes that this is the protocol that will be used (and perhaps adopted by other implementations), then configuring this would require parameters such as:

* Nominal bit time
* Angle value for a full rotation (up to 14 bits of the word are usable in the protocol as described)
* Number of extruder stepper steps for a full rotation
* Tolerance, as a total count of mm difference
* Tolerance, as a percentage difference between the two
* Recovery script G code sequences. (This isn't necessarily exactly the same as just firing up M600 to swap out filament when it runs out.)

I think it would be useful for the algorithm to trigger for accumulated error, but to "forgive" it if deltas continue to arrive that are themselves consistently within range.

thisiskeithb, 2020-10-07:
Marlin supports motion sensors. See `FILAMENT_MOTION_SENSOR`:

https://github.com/MarlinFirmware/Marlin/blob/c650b752f19040d3ee199b7c3c173d2dbe574da1/Marlin/Configuration.h#L1184-L1189

logiclrd, 2020-10-07:
That's not the same thing, though. That uses a 1-bit input that's basically following the heuristic, "If the value keeps changing, everything's good." There's no easy way to use that with e.g. Duet3D's Magnetic Filament Monitor. I suppose you could put an Arduino in between, have it decode the 16-bit signals, and then just output the value from bit 5 or something, but it's really not the feature requested here.

logiclrd, 2020-10-07:
Consider a partial blockage, where filament does keep moving, but only 50% of the requested steps actually translate into filament movement. The other 50% are eaten up by the hobbed gear skipping on the filament. With a simple encoder as supported by `FILAMENT_MOTION_SENSOR`, it's still going to see the signal toggling on and off multiple times within the `FILAMENT_RUNOUT_DISTANCE_MM` window, and won't know that anything is wrong. With the more advanced sensor, it can see, "Oh, I've told it to extrude 30mm of filament over the last _n_ steps, which means the wheel should have turned 500°, but it has only turned 250°."

Sebbb, 2020-11-18:
https://duet3d.dozuki.com/Wiki/Duet3dFilamentMonitor_LaserVersion uses a similar approach and protocol, having support for such open hardware would be nice.

superzerg, 2021-02-11:
There is support  in Marlin 2 for rotary encoders used for closed loop control including for extruder (look at I2C_POSITION_ENCODERS in Configuration_adv.h).
The code to read the encoder value will probably need to be modified to read this specific encoder, but the closed loop part is already here.

I am currently testing the closing of the loop with an AS5600 encoder on my Y axis with promising results, I will prepare a PR once I'm confident everything is Ok. Implementing the code for this new encoder took me  about a day so my guess is that it should not be quite easy to implement the one you linked to.

PS: I would not really recommend using the AS5600 sensor as its I2C address is fixed so a single encoder can be connected, but as I had one already, I wanted to use it.