MarlinFirmware/Marlin #19660: E0_AUTO_FAN_PIN on BTT SKR V1.4

killrob, 2020-10-07:
Hi all, is possible to have working E0_AUTO_FAN_PIN on a Bigtreetech SKR V1.4 to start the extruder fan over a specified temp?
Thank you

ellensp, 2020-10-07:
you simply set the E0_AUTO_FAN_PIN like any board.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-07 by ellensp

ellensp, 2020-10-08:
FAN0 is the only software controllerable fan on the skr 1.4 controller. This is a hardware limitation.