MarlinFirmware/Marlin #19662: [BUG] (short description)

borsegbr, 2020-10-09:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-10-09:
Please attach your configuration files and fill out the template

thisiskeithb, 2020-10-09:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/19602

closed 2020-10-09 by thisiskeithb