MarlinFirmware/Marlin #19666: [FR] Laser Functions

buzkill420, 2020-10-09:
##intro

Seems that Marlin isn't turning away from enabling users to use lasers on their machines, and i know grbl is out there, but the fact remains that Marlin has plenty to offer in the laser cutter/engraving space. It dominates over grbl in the LCD/user control department. (hats off to the team btw), already has a great temperature PID system with multiple inputs and settings available, multiple fan control, enclosure control, easy PWM frequency setting (important for powering tech cooler with PWM). 

So I hope I'm not annoying anyone by posting these possible features that could be added to make the laser set up that much "cooler." pun intended. And if there are already features similar or exactly what i describe please don't hesitate to tell me off! I've looked quite a lot and have not found much on this front, though many features of Marlin come close, lots of things disable settings when say no bed or extruder is included in the system configs. My hope is maybe some of these ideas could be implemented easily with just a few tweaks by  the team.

## Description

1. Laser Temp Probe and Tech cooling-  Some lasers come with Thermoelectric cooling and a temperature probe but no means of controlling the tech and fan.  With high frequency PWM, it's possible to vary the power to the unit with minimal efficiency losses and thus have a pretty simple means of controlling the laser temperature. Using one of the unused mosfet outputs on a RAMPS for example, could power a 60W Tech easily. Adding a second and third thermistor for the Tech hot-side, and a "room-temp" reading would allow for:
        A) a Laser and tech Over/Under-temp warning/killswitch/pause-until-safe-feature. 
        B) auto cooling  of the laser after a finishing a print by keeping the tech on until the laser has                         reached room temp.
        C) Thermal throttling of the tech if needed and safe to do so. (ie. tech is getting hot but laser is at safe temp)
 
2. Multiple Fans with PWM- Enclosure fan, Laser Fan, Tech Heat-sink Fan are just some of the possible fans used with a laser. 

3. ARMED/Disarmed Pin- An arm switch/key is pretty standard with most lasers for safety. Keys off, even if software says arm, nothing happens. Now it's trivial to accomplish with hardware, but it would be nice if  Marlin had an input pin to assign to ARMED_STATUS so that we could have a  status for ARMED/Disarmed on info screen.

4. Air-assist Menu item- Adding a Menu item and digital pin to enable a relay to turn on and off a pump for an Air-Assist system.

5. Info screen layout for Laser/ Cooler- the ability to see  Laser and Tech hot-side temps alongside laser amperage, Tech %. I know marlin has implemented a voltage and amperage sensor system, but getting that info onscreen would be nice.

### Feature Workflow

Please see videos.

#### Additional Information
Videos showing my attempt at making such a setup with Marlin 1.1.9.1 before 2.0.X.
[Video 1](https://www.youtube.com/watch?v=KvaruLahG9g)
[Video 2](https://www.youtube.com/watch?v=s0VNyzXNGVw)

bugfixin, 2021-01-08:
Regarding #4 - Currently at least some software (lightburn) expects to be able to use the standard coolant M7 and M8 gcodes to turn air assist on and off and it works relatively normally. This gives control of at least one pin/relay for that purpose. One could also add custom user menu items for it but I agree it would be nice for a more "built-in" experience.

The Armed/Disarmed is what I'm really interested in - safety interlock capability is sorely missed imho. Doing it in pure hardware is best from a safety perspective, but I'd like marlin to at minimum be aware of it and stop executing motion commands, too. Right now I think the closest workaround is using a filament runout sensor pin instead.

thisiskeithb, 2022-06-29:
Since this is added in https://github.com/MarlinFirmware/Marlin/pull/22690, I'll close this FR.

closed 2022-06-29 by thisiskeithb