MarlinFirmware/Marlin #19667: G29 J does not respect bed boundaries

Drzet, 2020-10-09:
### Bug Description

While using UBL, G29 P1 stays within boundaries defined in Marlin, however, when I add G29 L1 -> G29 J the extruder is trying to reach a point beyond Y max

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5355932/configs.zip)
### Steps to Reproduce

1. G28
2. G29 P1 -> G29 P3 -> G29 S1
3. G29 L1 -> G29 J

With G29 J probe 3 points on bed within the mesh boundaries

**Actual behaviour:** [What actually happens]
Front left and front right points are probed, then probe tries to reach a third 

point beyond Y max resulting in error

boelle, 2020-10-13:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

przepka1985, 2020-10-15:
Check x_bed_sixe and y_bed_size in config.h
I was get same issue with Geeetech a10t with bed dimensions 235x235 but real movement its 225x225 and x in - 1  ymin - 7 but will be different for your printer.

Drzet, 2020-10-17:
I can confirm that with the latest bugfix this bug is fixed (or is it a new feature?).
Thanks

closed 2020-10-17 by Drzet