MarlinFirmware/Marlin #19677: [BUG] MKS Robin E3 hangs after PR #19614

ldursw, 2020-10-10:
### Bug Description

After applying the PR #19614 the printer gets stuck on the bootloader screen, Marlin does not start up.

I have updated the branch 2.0.x from 2.0.7 to the latest commit and noticed Marlin didn't boot. I've done a bisect to find the commit and found 25cf9c6 to be causing the regression.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5359199/Marlin.zip)
Printer: Ender-3 Pro
Board: MKS Robin E3

### Steps to Reproduce

1. Build Marlin with the commit applied
2. Flash the firmware using a Micro SD card

**Expected behavior:** Marlin boots up and works normally

**Actual behavior:** Board gets stuck at bootloader

#### Additional Information

[image]

oscar6502, 2020-10-10:
Same problem witch skrpro and reprap full graphic smart controller

rhapsodyv, 2020-10-10:
can you try commenting out this like: https://github.com/MarlinFirmware/Marlin/blob/25cf9c65e27c1b0b38d045a1d474c699b39bd21e/Marlin/src/inc/Conditionals_LCD.h#L231

thisiskeithb, 2020-10-10:
I can confirm that I'm seeing the same issues after https://github.com/MarlinFirmware/Marlin/pull/19614 was merged.

> can you try commenting out this like:

That didn't work for my SKR Pro/Full Graphic Smart Controller build I have on the bench.

rhapsodyv, 2020-10-10:
One more test: change all those 0x1F to 0xF 👍 https://github.com/MarlinFirmware/Marlin/blob/25cf9c65e27c1b0b38d045a1d474c699b39bd21e/Marlin/src/HAL/STM32/fastio.h#L54

https://github.com/MarlinFirmware/Marlin/blob/25cf9c65e27c1b0b38d045a1d474c699b39bd21e/Marlin/src/HAL/STM32/fastio.h#L55

https://github.com/MarlinFirmware/Marlin/blob/25cf9c65e27c1b0b38d045a1d474c699b39bd21e/Marlin/src/HAL/STM32/fastio.h#L61
https://github.com/MarlinFirmware/Marlin/blob/25cf9c65e27c1b0b38d045a1d474c699b39bd21e/Marlin/src/HAL/STM32/fastio.h#L62

oscar6502, 2020-10-10:
> I can confirm that I'm seeing the same issues after #19614 was merged.
> 
> > can you try commenting out this like:
> 
> That didn't work for my SKR Pro/Full Graphic Smart Controller build I have on the bench.

I'm sorry, I don't speak English, I use a translator

thisiskeithb, 2020-10-10:
> I'm sorry, I don't speak English, I use a translator

I'm working with @rhapsodyv on Discord, so a fix should be incoming shortly.

oscar6502, 2020-10-10:
> > I'm sorry, I don't speak English, I use a translator
> 
> I'm working with @rhapsodyv on Discord, so a fix should be incoming shortly.

Thank you, I appreciate your work and that of the community!

thisiskeithb, 2020-10-10:
See https://github.com/MarlinFirmware/Marlin/pull/19678 for the fix.

ldursw, 2020-10-10:
Confirmed #19678 fixed the issue for me.

closed 2020-10-10 by thisiskeithb

sjasonsmith, 2020-10-10:
I am very confused how fixing HAL/STM32 fixed E3 boards that use HAL/STM32F1. Two people have said it fixes it though, so I am at a loss...

thisiskeithb, 2020-10-10:
> I am very confused how fixing HAL/STM32 fixed E3 boards that use HAL/STM32F1. Two people have said it fixes it though, so I am at a loss...

After we talked, I wanted to figure it out as well.

The Robin E3 board can be...particular about SD cards sometimes and won't boot (It will after ~30 seconds, but appears to be stuck on the "boot" text screen).

I re-flashed https://github.com/MarlinFirmware/Marlin/commit/6fdaaf3d203002fb7b437e86cd3aaa80e5da77bd and the Robin E3 booted. 🤷‍♂️

I think we just happened to stumble into the bug & fix thanks to @oscar6502's comment.

Serhiy-K, 2020-10-14:
I tested `fastio.h` for `STM32` on my STM32F411 board with next changes:

`#if defined(STM32F0xx) || defined(STM32F1xx) || defined(STM32F3xx) || defined(STM32L0xx) || defined(STM32L4xx)`
`  #define** _WRITE(IO, V) do { \`
`    if (V) FastIOPortMap[STM_PORT(digitalPin[IO])]->BSRR = _BV32(STM_PIN(digitalPin[IO])) ; \`
`    else   FastIOPortMap[STM_PORT(digitalPin[IO])]->BRR  = _BV32(STM_PIN(digitalPin[IO])) ; \`
 ` }while(0)`
`  #define _READ(IO)             bool(READ_BIT(FastIOPortMap[STM_PORT(digitalPin[IO])]->IDR, _BV32(STM_PIN(digitalPin[IO]))))`
`  #define _TOGGLE(IO)           (FastIOPortMap[STM_PORT(digitalPin[IO])]->ODR ^= _BV32(STM_PIN(digitalPin[IO])))`
`#else`
`  #define _WRITE(IO, V)         (FastIOPortMap[STM_PORT(digitalPinToPinName(IO))]->BSRR = _BV32((digitalPinToPinName(IO) & 0xF) + ((V) ? 0 : 16)))`
`  #define _READ(IO)             bool(READ_BIT(FastIOPortMap[STM_PORT(digitalPinToPinName(IO))]->IDR, _BV32(digitalPinToPinName(IO) & 0xF)))`
`  #define _TOGGLE(IO)           (FastIOPortMap[STM_PORT(digitalPinToPinName(IO))]->ODR ^= _BV32(digitalPinToPinName(IO) & 0xF))`
`#endif`

For MCUs listed in the first line used `digitalPin[IO]` and for other - `digitalPinToPinName(IO)`. For bit selection used `digitalPinToPinName(IO) & 0xF`.  `digitalPinToPinName (IO)` is required because in some `variant.h` files **A0-An** pins are defined as analog.

PR #19678 (2020-10-10): Fix FastIO for STM32 - Hal is broken
### Description

Go back to the original correct code. In the STM documentation: 

High nibble = port number (FirstPort <= PortName <= LastPort)
Low nibble  = pin number. 

So it must be 0xF, no 0x1F.

### Benefits

Fix #19677

### Related Issues

#19677 
#19614