MarlinFirmware/Marlin #19679: Bltouch do not stop axis Z

BeppeTemp, 2020-10-10:


yysh12, 2020-10-10:
have you tried #define USE_PROBE_FOR_Z_HOMING? 
Also it's probably good practice to raise z halfway up to max and then use your finger to test if it stops so you have enough time to cut power before it pierces the plate.

yysh12, 2020-10-10:
I also see you have
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
enabled which should enable USE_PROBE_FOR_Z_HOMING automatically

How is your bltouch wired up? Is it actually wired to the Z_min_endstop_pin? Or is it wired to separate pin? If there is a mismatch this could explain the problem.

BeppeTemp, 2020-10-10:
Yes, It Is wired to the Z_min_endstop_pin.

yysh12, 2020-10-10:
I see you have CONFIG_EXAMPLES_DIR set to "Creality/Ender-3/CrealityV1", which I believe means that you are using a default config file that was set up for the creality 1.1.x mainboard, yet you are using a third party mainboard from skr. I really think you have simply set z_probe to use the wrong pin within your config file.

yysh12, 2020-10-10:
oh wait never mind. You have the correct board set with define motherboard to BTT_SKR_V1_3. In that case I'm out of ideas. Next step I guess would be to use gcodes to manually verify if the printer is reporting the correct values for your sensors. idk.

BeppeTemp, 2020-10-10:
could you write me the g-code so tomorrow morning I'll try it please?

yysh12, 2020-10-10:
which bltouch are you using?

BeppeTemp, 2020-10-10:
3.0 thanks. I will see tomorrow.

yysh12, 2020-10-10:
as for gcode, just try one of the following from https://marlinfw.org/docs/gcode/:
M43 - Debug Pins 
M119 - Endstop States 
M226 - Wait for Pin State

ellensp, 2020-10-11:
the black/white cable has polarity, you must plug it in correct way. White needs to be on the signal pin

closed 2020-10-11 by ellensp