MarlinFirmware/Marlin #19688: [BUG] SKR PRO not boot with last stable version (need confirmation)

BastR, 2020-10-11:
### Bug Description

SKR PRO no boot since upgrading marlin to last stable version.

LCD is blue
The motors only move in one direction, if I ask X to go left and then right, the motor always goes left.
Fan are working
Octoprint can comunicate with the board
I didn't try to heat the nozzle or the bed.

### My Configurations

Functional config: https://github.com/BastR/Marlin/tree/Titan
No funcional config: https://github.com/BastR/Marlin/tree/Titan_NEW

I first tried to update my current firmware but it didn't work, so I'm back on a virgin marlin base with the latest stable version, it didn't work either.

I took a firmware available on the Bigtreetech github and the printer worked again.

I have a second printer with a similar setup, I used this base by updating the configuration files and everything works again...

So I can say that everything was functional on 30/09, since then something has been introduced and completely crashes the card.
[image]

 **I've done a lot of tests, someone would have to confirm what I'm saying because I get lost with all the things I've done.**

ellensp, 2020-10-11:
If it didn't boot, you wouldn't be able to talk to it at all...

rhapsodyv, 2020-10-11:
Can you try bugfix?

Patag, 2020-10-11:
SKR pro boots fine but the issue is LCD related. Latest (today) bugfix fixed it

BastR, 2020-10-11:
Vraiment pas de chance...

It now works with the current bugfix

closed 2020-10-11 by BastR