MarlinFirmware/Marlin #19691: [BUG] Anycubic Predator config based on the example stuck on Status screen and doesn't continue to info screen

NuClearArrow112, 2020-10-11:
#3766 # Bug Description

I downloaded the latest 2.0.7.1 bugfix branch and the bugfix example configurations. I made some minor modifications to configuration.h and configuration.h also added a custom bootscreen (not mine). I used an up to date VS Code with Auto Build Marlin. I was able to successfully compile but with these (I guess) warnings (yellow lines):

C:\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\STM32ADC\src\utility\util_adc.c:10:30: warning: 'adc_result' initialized and declared 'extern' extern volatile unsigned int adc_result = 0;
C:\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\FreeRTOS701\utility\port.c: In function 'xPortStartScheduler':
C:\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\FreeRTOS701\utility\port.c:178:2: warning: implicit declaration of function 'systick_attach_ca
.pio/build/trigorilla_pro/src/src/MarlinCore.cpp.o: In function `se.pio/build/trigorilla_pro/src/src/MarlinCore.cpp.o: In function `setup()':
C:\Marlin 2.0.7.1 Anycubic Predator/Marlin\src/MarlinCore.cpp:1094: warning: undefined reference to `TouchButtons::init()'
.pio/build/trigorilla_pro/src/src/lcd/ultralcd.cpp.o: In function `MarlinUI::update()':
C:\Marlin 2.0.7.1 Anycubic Predator/Marlin\src\lcd/ultralcd.cpp:888: warning: undefined reference to `TouchButtons::read_buttons()'

I tried to upload it to the board, and that was successfull as well, custom bootscreen worked, status screen worked, but it did not continue the info screen. The touchscreen doesn't work. I tried to connect to it with Pronterface but that didn't work. Please hep me, I have absolutely what did I do wrong.

### My Configurations

[bug submit.zip](https://github.com/MarlinFirmware/Marlin/files/5361888/bug.submit.zip)

1. Compile with my config files
2. Upload the bin file to a Trigorilla Pro board with its screen
3. Experience the issue as it's turned os

**Expected behavior:**

Marlin boots and show the info screen after the custom bootscreen and status screen

**Actual behavior:**

Stuck on the status screen, the touchscreen doesn't work, cannot communicate with it via USB.

sjasonsmith, 2020-10-11:
@NuClearArrow112 I posted a pull request to fix this yesterday, but closed it because it overlaps with other TFT work that is in-progress.

You can use the change from this closed PR to fix that read_buttons warning for now.
#19683 

It will be fixed in `bugfix-2.0.x` whenever the following larger PR is merged:
#19192

NuClearArrow112, 2020-10-12:
So the solution is patience. Well, I have a lot of that. Thank you very much!

rhapsodyv, 2020-10-12:
It's fixed in bugfix. @NuClearArrow112 can you test it?

Note that the Configuration got updated!

Thanks!

NuClearArrow112, 2020-10-13:
Still has the following warnings:

In file included from Marlin\src\lcd\tft_io\tft_io.cpp:30:0:
Marlin\src\lcd\tft_io\r65105.h:54:1: warning: multi-line comment [-Wcomment]
 // #define R61505_ETMOD_ORIENTATION  IF_0((TFT_ORIENTATION) & TFT_EXCHANGE_XY, R61505_ETMOD_AM) | \
 ^
In file included from Marlin\src\lcd\tft_io\tft_io.cpp:31:0:
Marlin\src\lcd\tft_io\ili9328.h:52:1: warning: multi-line comment [-Wcomment]
 // #define ILI9328_ETMOD_ORIENTATION IF_0((TFT_ORIENTATION) & TFT_EXCHANGE_XY, ILI9328_ETMOD_AM) | \
 ^

C:\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\STM32ADC\src\utility\util_adc.c:10:30: warning: 'adc_result' initialized and declared 'extern'
 extern volatile unsigned int adc_result = 0;
                              ^~~~~~~~~~

C:\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\FreeRTOS701\utility\port.c: In function 'xPortStartScheduler':
C:\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\FreeRTOS701\utility\port.c:178:2: warning: implicit declaration of function 'systick_attach_callback' [-Wimplicit-function-declaration]
  systick_attach_callback(&xPortSysTickHandler);
At top level:
C:\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\FreeRTOS701\utility\port.c:285:6: warning: 'prvSetupTimerInterrupt' defined but not used [-Wunused-function]
 void prvSetupTimerInterrupt( void )
      ^~~~~~~~~~~~~~~~~~~~~~

HOWEVER, it works like a charm, I tested the TFT rotation 180° as well, no problems there either.
Thank you very much again!
The only minor inconvenience is when I rotate the TFT the touchscreen doesn't rotate with it. That means the buttons are at the bottom of the screen but I have to touch on the top.
Video link, hope it works: [https://www.facebook.com/100003510221961/videos/3160021507458159/](url)
And it's not that intuitive how to do that by these lines:

#define XPT2046_X_CALIBRATION   12316
#define XPT2046_Y_CALIBRATION    -8981
#define XPT2046_X_OFFSET                  -43
#define XPT2046_Y_OFFSET                 257

But I can rotate the whole display physically, so problem solved. Thanks.

rhapsodyv, 2020-10-13:
Try this: https://github.com/MarlinFirmware/Marlin/pull/19710

sjasonsmith, 2020-10-26:
You will also need the change from this PR (in the config repo) for the Predator example:
https://github.com/MarlinFirmware/Configurations/pull/288

I haven't actually tested that configuration change. but at least it compiles. Feedback about whether or not it works is welcome.

sjasonsmith, 2020-11-16:
This issue has been silent, so I am presuming it has been fixed.

We have another Predator user who said there was still another issue and added a new PR to fix the screen rotation:
https://github.com/MarlinFirmware/Configurations/pull/328

Since it seems the only outstanding issue was screen rotation, I'm presuming the reported issue here is resolved, and will close this.

closed 2020-11-16 by sjasonsmith