MarlinFirmware/Marlin #19692: [FR] Bed levelling for CNC router

Tall-Paul, 2020-10-11:
### Bug Description

It currently doesn't seem as if the various bed levelling options (although to be honest I've only been able to test auto bi-linear) don't work when your Z-moves are below the 'bed' (ie: levelling is done to a workpiece, and all moves go below this surface).

I'm not sure however whether this is because of a bug in the levelling, or because the levelling was never designed to work in this way.

I've added an image of a PCB I milled out on a modified Anet A6 with a BLTouch sensor and a spindle mount.  I've run this multiple times, issuing a G29 before each run and the height is off by the same amount in the same X,Y position each time which (to me) indicates that the levelling isn't working correctly rather than my probe having problems with repeatability or whatever.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5361993/Configuration.zip)

### Steps to Reproduce

Build Marlin with attached config
Deploy to a 'printer' modified with a bltouch sensor and a spindle mount.

G28
G29
M420 S1

Run Gcode file

**Expected behavior:** bed levelling is used

**Actual behavior:** bed levelling doesn't seem to be used

#### Additional Information

Image attached showing result of a milled PCB.  Front-right shows cuts at the correct depth.  left / back of the board are clearly too shallow.

[image]

sjasonsmith, 2020-10-12:
This has come up before, but was closed without resolution: #18072

Are you configuring your mill to prove only over the area you will be cutting, or are you probing across the entire bed area (including off the PCB!). I am wondering if there are some steps missing in your "Steps to reproduce" related to setting the zero points for your milling job.

The following are some very generic leveling debugging instructions. These generally assume printing...namely I don't know if you will be able to do "Do some of the moves that revealed problems", since the PCB would be in the way. Perhaps after probing you could remove the PCB, lower your Z slightly below 0, then move around.

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin bugfix-2.0.x to test with the latest code.
- Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send M502 and M500 to ensure your Configurations are applied.
- Issue the command M111 S247 to enable maximum logging.
- Perform a G28 to do your standard homing procedure.
- Do a G29 to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a .TXT file and attach it to your next reply.
- Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Tall-Paul, 2020-10-12:
I'm configuring the mill to have a bed the same size as the blank board, and my 0,0 is the bottom left of the board.  The probing routine correctly covers the board (and only the board) and I do 5*5 points.

I'm actually struggling to build the bugfix branch, the 2.0x branch builds fine but when building bugfix I get an error relating to the u8g library and the LCD pins not being declared in fastio (or something like that, not sat at the build computer right now).  As soon as I can actually get a working build of bugfix I'll follow the steps above. 

I guess I just wanted to check if the levelling is designed to work below z=0 or if the assumption was made that it would only be required above zero before I started putting too much effort into it.

Tall-Paul, 2020-10-17:
Okay, managed to get the bugfix branch compiled.  I also did some work on stiffening up my machine, reducing play wherever I can etc.

Note in the log I did G29 twice to check repeatability. 

The bugtest.cnc file is the G-code I used to produce the line on the right of this picture.  It barely scratches the surface from Y10 to Y20, then almost immediately starts to cut deeper. 

The circuit in the rest of the picture was cut with the same levelling data (ie: after I did the levelling and did the log) but I didn't save the log for that.  I've attached the G-code for the circuit as well in case it's any use

[image]

[log.txt](https://github.com/MarlinFirmware/Marlin/files/5396467/log.txt)

[bugtest.cnc.txt](https://github.com/MarlinFirmware/Marlin/files/5396468/bugtest.cnc.txt)

[deskpcb-F_Cu.gbr_1_0.3000_iso_cnc.txt](https://github.com/MarlinFirmware/Marlin/files/5396518/deskpcb-F_Cu.gbr_1_0.3000_iso_cnc.txt)