MarlinFirmware/Marlin #19700: Using (BL|3D)TOUCH on a stock Ender 3 mobo on PIN 29 with BEEPER

459below, 2020-10-12:
Isn't this hardcoded for BLTOUCH to use PIN 27?

https://github.com/MarlinFirmware/Marlin/blob/e4c23f85d39228cf6386d80927e33af0769e751e/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h#L71

I connected PIN 29 to preserve beeper functionality like shown in this thing https://www.thingiverse.com/thing:3293821
[image]

Changing it from 27 to 29 gets BLTOUCH working, however commenting out

```
#undef BEEPER_PIN
```

makes the firmware not booting most of the time. The screen stays blank entirely, still in some circumstances it does come up and then the beeper works as well as the sensor.

I seem to be missing something about the BLTOUCH option influencing the SOUND or BEEPER options. 🤔

cguerrero1205, 2020-10-13:
Hi, I didn't understand your question. Which pin is the buzzer connected to and which pin is the BLT connected to? It would be good if you share the files you modified.

459below, 2020-10-13:
Hi,
well, sorry for being less than clear.

So the buzzer is connected to PIN 27. Just the unmodified stock cable, no PIN 27 board or splicing there.

My 3DTOUCH  is connected to PIN 29 after soldering in a header to the unpopulated hole like in the picture.

I really just modified this one line in the entire firmware. This is booting but without the BEEPER, even though it is still connected.

```
diff --git a/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h b/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h
index 9f406c1f7..3002ed423 100644
--- a/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h
+++ b/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h
@@ -68,7 +68,7 @@
 #define LCD_PINS_D4                          30   // ST9720 CLK
 
 #if ENABLED(BLTOUCH)
-  #define SERVO0_PIN                         27
+  #define SERVO0_PIN                         29
   #undef BEEPER_PIN
 #elif ENABLED(FILAMENT_RUNOUT_SENSOR)
   #ifndef FIL_RUNOUT_PIN
```

But when I would comment out _#undef BEEPER_PIN_ the board doesn't boot properly.

Here's the gist of my modifications in the configuration:
https://gist.github.com/459below/a1f1c4d8fdb69f15b62df0b8de006671

boelle, 2020-10-13:
is this a bug or question?

459below, 2020-10-13:
Yeah mostly a question. The question being:
Can one enable BLTOUCH and BEEPER simultaneously when using PIN 29 instead of PIN 27 for BLTOUCH?

qwewer0, 2020-10-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: [https://discord.gg/n5NJ59y](https://discord.gg/n5NJ59y)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-13 by 459below