MarlinFirmware/Marlin #19702: Did not compile when enable Mechanical Gantry Calibration

Talha909, 2020-10-12:
I have download the New bug fix . And enable the Mechanical Gantry Calibration. But when i enable
[image]
 this then frimware did not compile. I got error message.

ellensp, 2020-10-12:
Please attach your configuration files.

Talha909, 2020-10-12:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5364660/Marlin.zip)
i have attached the config files. I had installed the SKR 1.4 TURBO with TMC2130 Drivers.

Talha909, 2020-10-12:
> Please attach your configuration files.

Files are attached. I had installed SKR 1.4 TURBO with TMC 2130 and BLtouch.

ellensp, 2020-10-12:
you have Z_STEPPER_AUTO_ALIGN and MECHANICAL_GANTRY_CALIBRATION
these seem to be mutually exclusive, since both define G34

Talha909, 2020-10-12:
> you have Z_STEPPER_AUTO_ALIGN and MECHANICAL_GANTRY_CALIBRATION
> these seem to be mutually exclusive, since both define G34

What is mechanical gantry calibration.?

ellensp, 2020-10-12:
introduced here https://github.com/MarlinFirmware/Marlin/pull/18972

ellensp, 2020-10-12:
so replaces https://marlinfw.org/docs/gcode/M915.html

Talha909, 2020-10-12:
I disable the Align Z and its compile now.

thisiskeithb, 2020-10-12:
I created a sanity check in https://github.com/MarlinFirmware/Marlin/pull/19706 to prevent compiling when both options are enabled.

closed 2020-10-12 by thinkyhead

Talha909, 2020-10-13:
Hi, 

Today I test it . But when I gave the command of G34 it always hit the bed and did not stop.  I am using Bltouch