MarlinFirmware/Marlin #19704: [marlin 2.0.7] (no movement at all on any steppers)

bywebberen, 2020-10-12:
Hey there, I recently updated the marlin on my DADSON PRUSA I3 PRINTER - it worked before, but I wanted the newest version, to try some of the new features. My board is the Bigtreetech SKR v1.3 - and it was running great with the older version of marlin. After I have changed to newest, I have no movement on any steppers at all...?

closed 2020-10-12 by ellensp

ellensp, 2020-10-12:
I will however add this is often the result of people enabling #define SOFTWARE_DRIVER_ENABLE withing knowing if their stepper driver uses this feature or not. (most do not)

bywebberen, 2020-10-12:
I will try thanks.. I use TMC2209

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>