MarlinFirmware/Marlin #19707: [FR] Save the Level Data somewher else?

FlailAway, 2020-10-12:
After a bed level and M500 the co-ords are saved, all good. But, the next upload of marlin 2.xxx will drop it all no matter how trivial the change to the Firmware was.

Is there some way to prevent this? I just changed the new screen selection to move Right for Encoder Clockwise and bang, all level data has gone and pretty sure the bed hasn't moved. Now I sit while it does a level again. Could it be store on the SD card? Or, somewhere else and replaced in the EEPROM after a Firmware update? Maybe temporarily held in Program Storage Space when updating.

Vertabreak, 2020-10-12:
this kind of already exists via the use of G29 S-1 when using UBL it plays out the mesh data in gcode so you can use it to load the mesh data back in from external gcode.

thinkyhead, 2020-10-12:
There is already a project posted to refactor the EEPROM so that all data would be preserved. We don't want to do this piecemeal and waste time we could be spending here: https://github.com/MarlinFirmware/Marlin/projects/3

FlailAway, 2020-10-13:
> this kind of already exists via the use of G29 S-1 when using UBL it plays out the mesh data in gcode so you can use it to load the mesh data back in from external gcode.

I tried using that but got a "mesh does not exist" error. Could not find how to use it correctly.