MarlinFirmware/Marlin #19709: MKS Robin nano v.2 bootloop

Malderin, 2020-10-12:
Marlin 2.0.Bugfix.
After update:
https://github.com/MarlinFirmware/Marlin/pull/19192
the board is rebooted upon loading after the start logo. Next comes the endless reboot (bootloop).
File configuration.h  set parameters:
#define MKS_TS35_V2_0
#define TFT_LVGL_UI

tpruvot, 2020-10-12:
was due to 0988af453c567fae1796cc821bbc41e930b45836

may have been fixed today with tft refactor commit

Malderin, 2020-10-13:
Marlin 2.0.7.2 (stable).
File configuration.h set parameters:
#define MKS_TS35_V2_0
#define TFT_LVGL_UI
The board boots, but the color gamut is incorrect. Tried adding parameters:
#define TFT_COLOR TFT_COLOR_RGB
 #define TFT_COLOR TFT_COLOR_BGR
to configuration.h, nothing changes.
See video: [error](https://yadi.sk/d/4D6h30Yrq1nwAg)
My fork with configurations: https://github.com/Malderin/Marlin_2.0.x_official/tree/Marlin2.0.7

Malderin, 2020-10-13:
after rolling back this commit:
https://github.com/MarlinFirmware/Marlin/commit/fea0db47d
board started normally.

rhapsodyv, 2020-10-13:
Please try #19710 . It's fixed now.

Thanks!

Malderin, 2020-10-13:
> Please try #19710 . It's fixed now.
> 
> Thanks!

1. Marlin 2.0.7.2 problem solved
2. Marlin 2.0.bugfix, the color scheme has become normal (visible on the splash screen), but the bootloop is still

rhapsodyv, 2020-10-13:
What are your configs?

I need to disable "TOUCH_SCREEN" when using LVGL.

Malderin, 2020-10-13:
> What are your configs?
> 
> You need to disable "TOUCH_SCREEN" when using LVGL.
Found the reason. If #define USE_WATCHDOG is uncommented in Configuration_adv.h then the board goes into reboot.

My configurations:
[Conf.zip](https://github.com/MarlinFirmware/Marlin/files/5373397/Conf.zip)
or in my fork:
https://github.com/Malderin/Marlin_2.0.x_official/tree/Marlin2.0.7_bugfix

Malderin, 2020-10-13:
> The watchdog problem is fixed here: #19724

the fix did not help, the bootloop was preserved

rhapsodyv, 2020-10-13:
I tested *your* configs in my Nano V2 + MKS TS 35, using #19724, it boot normally. 100%

th0m4sek, 2020-10-14:
Problem is still there and it's seems to be related to watchdog. My MKS robin nano 1.2 is not starting every time. With new bug fix problem is still present. I've commented #define USE_WATCHDOG and it's booting.

Malderin, 2020-10-14:
> Problem is still there and it's seems to be related to watchdog. My MKS robin nano 1.2 is not starting every time. With new bug fix problem is still present. I've commented #define USE_WATCHDOG and it's booting.

I confirm, I still have a problem with active USE_WATCHDOG

rhapsodyv, 2020-10-14:
@makerbase-mks is possible of some nano boards came with the hardware watchdog enabled? Or some bootloader version turning watchdog on before marlin start?

rhapsodyv, 2020-10-14:
> > Problem is still there and it's seems to be related to watchdog. My MKS robin nano 1.2 is not starting every time. With new bug fix problem is still present. I've commented #define USE_WATCHDOG and it's booting.
> 
> 
> 
> I confirm, I still have a problem with active USE_WATCHDOG

The problem is with nano v1.2 or nano v2?

rhapsodyv, 2020-10-14:
Here, both nano v1.2 and nano v2 with bugfix + watchdog enabled. 

Or we are testing different marlin code base, or MKS have different setups for the same board...

[image]
[image]

Malderin, 2020-10-14:
> Here, both nano v1.2 and nano v2 with bugfix + watchdog enabled.
> 
> Or we are testing different marlin code base, or MKS have different setups for the same board...
> 
> [image]
> [image]

Oк. You can test this firmware for Robin nano v.2? 
This is compiled from the latest bugfix sources and causes a bootloop on my board.

[Robin_nano35.zip](https://github.com/MarlinFirmware/Marlin/files/5378707/Robin_nano35.zip)

rhapsodyv, 2020-10-14:
> [Robin_nano35.zip](https://github.com/MarlinFirmware/Marlin/files/5378707/Robin_nano35.zip)

Your firmware reboot my board. So, the problem is not the board. Seems your marlin codebase is outdated.

Can you test with a "fresh" copy and build?

Try this firmware that I compiled now, for nano v2:

[Robin_nano35.bin.zip](https://github.com/MarlinFirmware/Marlin/files/5378739/Robin_nano35.bin.zip)

Malderin, 2020-10-14:
@rhapsodyv , I found where I had the problem that was causing the bootloop.
In file Marlin\src\lcd\extui\lib\mks_ui\SPI_TFT.cpp in section:
void TFT::LCD_init() {
  tftio.InitTFT();
  #if PIN_EXISTS(TFT_BACKLIGHT)
    OUT_WRITE(TFT_BACKLIGHT_PIN, LOW);
  #endif
  delay(100);
  LCD_clear(0x0000);
  LCD_Draw_Logo();
  #if PIN_EXISTS(TFT_BACKLIGHT)
    OUT_WRITE(TFT_BACKLIGHT_PIN, HIGH);
  #endif
  #if HAS_LOGO_IN_FLASH
    delay(2000);
  #endif
}
Instead of delay (2000) I had delay (4000) specified. I returned 2000 and the bootlap was gone. It's strange ...

th0m4sek, 2020-10-14:
I'm suspecting some problems with my microsd card because firmware upload was also not always success. I used different card this time and i will observe printer.