MarlinFirmware/Marlin #19711: [BUG] 2.0.7.1 won't boot on SKR E3 V2.0

gitmiz, 2020-10-13:
### Bug Description

Merged my origin (2.0.6.1) with Marlin 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5368503/Configuration.zip)
upstream 2.0.x (2.0.7.1) and it no longer boots.  Accepted all incoming changes.  Reverted back to 2.0.6.1 via git reset --hard to unmerge and it works fine again.  Something is preventing 2.0.7.1 from booting on the SKR E3 2.0.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Flash 2.0.7.1 on SKR Mini E3 V2.0

**Expected behavior:** Successful boot

**Actual behavior:** Board never fully boots and LCD stays blank even though firmware is successfully written.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-10-13:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

gitmiz, 2020-10-13:
> Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

The latest bugfix-2.0.x boots.  The only configuration changes made on that build is changing the serial ports and enabling CR10 stock display.

rhapsodyv, 2020-10-13:
So is it fixed?

gitmiz, 2020-10-13:
> So is it fixed?

I don't know.  2.0.7.1 definitely not, so maybe something was pulled into bugfix that resolves this.  The only way to be certain is for me to duplicate all of the configurations in my bugfix fork.

rhapsodyv, 2020-10-13:
you said:  "The latest bugfix-2.0.x boots."....

gitmiz, 2020-10-13:
> you said: "The latest bugfix-2.0.x boots."....

The bug was opened for Marlin 2.0.7.1, not bugfix.  The attached configurations work in 2.0.6.1 and not in 2.0.7.1.  Nevermind..

rhapsodyv, 2020-10-13:
If it's fixed in the bugfix, then it is fixed... because, well, bugfix is for that: bug fixes... 😄 

Soon, bugfix will be merged into the main branch.

closed 2020-10-13 by thisiskeithb