MarlinFirmware/Marlin #19714: Display Driver for CR20-Pro wont let it compile

MartianNinja, 2020-10-13:
### Bug Description

### My Configurations
[CR-20 Pro.zip](https://github.com/MarlinFirmware/Marlin/files/5368951/CR-20.Pro.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Copy the 3 files into the marlin 2.0.x folder
2. I enabled Advanced pause, a 7x7 mesh and nozzle park feature
3. try to compile

**Expected behavior:** [What you expect to happen]
It should compile and give me a hex file which i can flash
**Actual behavior:** [What actually happens]
It gives me an error, an image of the error is attached. 
[image]

#### Additional Information
This has been confirmed by Discord user/dev sjasonsmith and also Dust.
I was also told by sjasonsmith that the below commit is the commit that broke it:
https://github.com/MarlinFirmware/Marlin/commit/193c0a52d9f1fbd31e42a29734f551aa26ed269a

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-10-13:
The error above means you're missing the `_Statusscreen.h` file, but I compiled your configs and get a different error:

```prolog
In file included from Marlin/src/lcd/menu/menu_motion.cpp:31:0:
Marlin/src/lcd/menu/menu_motion.cpp: In function 'void menu_move()':
Marlin/src/lcd/menu/menu_motion.cpp:222:58: error: attempt to take address of bit-field structure member 'soft_endstops_t::<anonymous struct>::_enabled'
     EDIT_ITEM(bool, MSG_LCD_SOFT_ENDSTOPS, &soft_endstop._enabled);
                                                          ^
Marlin/src/lcd/menu/menu_item.h:259:39: note: in definition of macro '_MENU_INNER_P'
     MenuItem_##TYPE::action(plabel, ##V);                      \
                                       ^
Marlin/src/lcd/menu/menu_item.h:354:57: note: in expansion of macro '_MENU_ITEM_P'
 #define MENU_ITEM_P(TYPE, PLABEL, V...)                 _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
                                                         ^
Marlin/src/lcd/menu/menu_item.h:392:58: note: in expansion of macro 'MENU_ITEM_P'
 #define EDIT_ITEM_P(TYPE, PLABEL, V...)                  MENU_ITEM_P(TYPE, PLABEL, ##V)
                                                          ^
Marlin/src/lcd/menu/menu_item.h:393:58: note: in expansion of macro 'EDIT_ITEM_P'
 #define EDIT_ITEM(TYPE, LABEL, V...)                     EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                          ^
Marlin/src/lcd/menu/menu_motion.cpp:222:5: note: in expansion of macro 'EDIT_ITEM'
     EDIT_ITEM(bool, MSG_LCD_SOFT_ENDSTOPS, &soft_endstop._enabled);
     ^
Marlin/src/lcd/menu/menu_motion.cpp:222:58: error: attempt to take address of bit-field structure member 'soft_endstops_t::<anonymous struct>::_enabled'
     EDIT_ITEM(bool, MSG_LCD_SOFT_ENDSTOPS, &soft_endstop._enabled);
                                                          ^
Marlin/src/lcd/menu/menu_item.h:264:58: note: in definition of macro '_MENU_INNER_P'
       (encoderLine == _thisItemNr, _lcdLineNr, plabel, ##V);   \
                                                          ^
Marlin/src/lcd/menu/menu_item.h:354:57: note: in expansion of macro '_MENU_ITEM_P'
 #define MENU_ITEM_P(TYPE, PLABEL, V...)                 _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
                                                         ^
Marlin/src/lcd/menu/menu_item.h:392:58: note: in expansion of macro 'MENU_ITEM_P'
 #define EDIT_ITEM_P(TYPE, PLABEL, V...)                  MENU_ITEM_P(TYPE, PLABEL, ##V)
                                                          ^
Marlin/src/lcd/menu/menu_item.h:393:58: note: in expansion of macro 'EDIT_ITEM_P'
 #define EDIT_ITEM(TYPE, LABEL, V...)                     EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                          ^
Marlin/src/lcd/menu/menu_motion.cpp:222:5: note: in expansion of macro 'EDIT_ITEM'
     EDIT_ITEM(bool, MSG_LCD_SOFT_ENDSTOPS, &soft_endstop._enabled);
Compiling .pio/build/mega2560/src/src/sd/SdFatUtil.cpp.o
     ^
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-register'
*** [.pio/build/mega2560/src/src/lcd/menu/menu_motion.cpp.o] Error 1
```

sjasonsmith, 2020-10-13:
You are right @thisiskeithb, that was just the incorrect screenshot accidentally included.

sjasonsmith, 2020-10-13:
This started failing because the endstop states changes from booleans to a bitfield, and those aren't usable in the menu macros.

thinkyhead, 2020-10-13:
For some reason in my testing - and CI - this path never got compiled…? I was just thinking the compiler was being clever… But of course not! Anyway, the bit field only saves 2 bytes of SRAM while it increases code size by some ~16 bytes. So, bit field here is of limited value. I'll change back to regular `bool`.

closed 2020-10-13 by thinkyhead

PioDes, 2020-10-13:
I get a similar error when I try to compile what should I do?

In file included from Marlin/src/lcd/menu/menu_motion.cpp:31:
Marlin/src/lcd/menu/menu_motion.cpp: In function 'void menu_move()':
Marlin/src/lcd/menu/menu_motion.cpp:222:58: error: attempt to take address of bit-field
  222 |     EDIT_ITEM(bool, MSG_LCD_SOFT_ENDSTOPS, &soft_endstop._enabled);
      |                                                          ^~~~~~~~
Marlin/src/lcd/menu/menu_item.h:259:39: note: in definition of macro '_MENU_INNER_P'
  259 |     MenuItem_##TYPE::action(plabel, ##V);                      \
      |                                       ^
Marlin/src/lcd/menu/menu_item.h:354:57: note: in expansion of macro '_MENU_ITEM_P'
  354 | #define MENU_ITEM_P(TYPE, PLABEL, V...)                 _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
      |                                                         ^~~~~~~~~~~~
Marlin/src/lcd/menu/menu_item.h:392:58: note: in expansion of macro 'MENU_ITEM_P'
  392 | #define EDIT_ITEM_P(TYPE, PLABEL, V...)                  MENU_ITEM_P(TYPE, PLABEL, ##V)
      |                                                          ^~~~~~~~~~~
Marlin/src/lcd/menu/menu_item.h:393:58: note: in expansion of macro 'EDIT_ITEM_P'
  393 | #define EDIT_ITEM(TYPE, LABEL, V...)                     EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
      |                                                          ^~~~~~~~~~~
Marlin/src/lcd/menu/menu_motion.cpp:222:5: note: in expansion of macro 'EDIT_ITEM'
  222 |     EDIT_ITEM(bool, MSG_LCD_SOFT_ENDSTOPS, &soft_endstop._enabled);
      |     ^~~~~~~~~
Marlin/src/lcd/menu/menu_motion.cpp:222:58: error: attempt to take address of bit-field
  222 |     EDIT_ITEM(bool, MSG_LCD_SOFT_ENDSTOPS, &soft_endstop._enabled);
      |                                                          ^~~~~~~~
Marlin/src/lcd/menu/menu_item.h:264:58: note: in definition of macro '_MENU_INNER_P'
  264 |       (encoderLine == _thisItemNr, _lcdLineNr, plabel, ##V);   \
      |                                                          ^
Marlin/src/lcd/menu/menu_item.h:354:57: note: in expansion of macro '_MENU_ITEM_P'
  354 | #define MENU_ITEM_P(TYPE, PLABEL, V...)                 _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
      |                                                         ^~~~~~~~~~~~
Marlin/src/lcd/menu/menu_item.h:392:58: note: in expansion of macro 'MENU_ITEM_P'
  392 | #define EDIT_ITEM_P(TYPE, PLABEL, V...)                  MENU_ITEM_P(TYPE, PLABEL, ##V)
      |                                                          ^~~~~~~~~~~
Marlin/src/lcd/menu/menu_item.h:393:58: note: in expansion of macro 'EDIT_ITEM_P'
  393 | #define EDIT_ITEM(TYPE, LABEL, V...)                     EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
      |                                                          ^~~~~~~~~~~
Marlin/src/lcd/menu/menu_motion.cpp:222:5: note: in expansion of macro 'EDIT_ITEM'
  222 |     EDIT_ITEM(bool, MSG_LCD_SOFT_ENDSTOPS, &soft_endstop._enabled);
      |     ^~~~~~~~~
Compiling .pio/build/LPC1769/src/src/libs/numtostr.cpp.o
*** [.pio/build/LPC1769/src/src/lcd/menu/menu_motion.cpp.o] E

Vertabreak, 2020-10-13:
this was fixed i merged and verified the change on my fork about an hour ago.