MarlinFirmware/Marlin #19715: MECHANICAL GANTRY CALIBRATION did not work well

Talha909, 2020-10-13:
I had enable the MECHANICAL_GANTRY_CALIBRATION. But when I gave the command of G34. The nozzel comes to middle and ignore the min position and hit the bed. I am using SKR 1.4 Tubo , TMC2130 and Bloutch 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5369379/Marlin.zip)

thisiskeithb, 2020-10-13:
I've only tested this on a MK3S with a fix-mounted probe, but I did not have this issue. I'll give it a test on my BLTouch printers in a bit to see if I can reproduce the issue.

Talha909, 2020-10-13:
[image]
this is my setting.

thisiskeithb, 2020-10-14:
Does your BLTouch work consistently?

I tried `MECHANICAL_GANTRY_CALIBRATION` on three different printers:

1. BTT002 with a PINDA V2/fix-mounted probe
2. SKR 1.3 with a BLTouch and drivers used in both single driver/dual Z & dual driver/independent Z motors
3. An LPC1769-based printer with a BLTouch and drivers used in both single driver/dual Z & dual driver/independent Z motors

They all work as expected running the latest `bugfix-2.0.x`.

There's now a sanity check preventing both `Z_STEPPER_AUTO_ALIGN` and `MECHANICAL_GANTRY_CALIBRATION` like you have enabled in your config, but since you're running independent drivers for Z, I'd recommend using `Z_STEPPER_AUTO_ALIGN`.

Talha909, 2020-10-14:
I download the Latest bug fix and disable the Z_Stepper auto align and enable the Mechanical Gantry Calibration. I go the same error. My Blotuch is connected on Z-Min.  Probe did not deploy when go down. It goes in center and going down and down and hit the bed.  Can you share you config files.  I have attached my config files. Z stepper align and Mechanical Gantry are both same thing or different. 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5376304/Marlin.zip)

thisiskeithb, 2020-10-14:
>  Probe did not deploy when go down. It goes in center and going down and down and hit the bed. 

Does it fail before moving the X gantry to the top? Does your printer home correctly with `G28` consistently? 

> Z stepper align and Mechanical Gantry are both same thing or different

They are different. `Z_STEPPER_AUTO_ALIGN` uses at least two independent Z drivers/steppers & a probe to align your X gantry to the bed (most accurate). `MECHANICAL_GANTRY_CALIBRATION` can work with either two Z steppers tied to a single driver or independent Z drivers/steppers and runs your X gantry to the top of your Z travel limit + the defined extra height, similar to how the Prusa MK3/S levels the gantry (less accurate).

[Here's my BTT002 (PINDA V2) config](https://github.com/thisiskeithb/Marlin/tree/myconfig/MK3S-BTT002-TMC2209)
[Here's my SKR 1.3 (BLTouch) config](https://github.com/thisiskeithb/Marlin/tree/myconfig/AM8-Bear-SKR13-TMC22209-BMG-Volcano) - `Z_STEPPER_AUTO_ALIGN` is currently enabled, but I tested with `MECHANICAL_GANTRY_CALIBRATION`

On my BLTouch-based printers, the probes deploy as they should, so I don't think there's in inherent problem with the procedure and likely something machine/wiring-specific.

thisiskeithb, 2020-10-14:
Wait. Why do you have `GANTRY_CALIBRATION_TO_MIN` enabled? Comment that out and try the process again.

Talha909, 2020-10-15:
I like the most accurate . I will use Z_stepper auto align.

closed 2020-10-15 by Talha909