MarlinFirmware/Marlin #19716: [FR] Require nozzle to be heated up before Z homing / probing

hamster65, 2020-10-13:
Sometimes, there is blob a plastic on the nozzle. When homing or probing with a cold nozzle, this can hit the bed before the z endstop / probe triggers. Heating up the nozzle would solve that.

qwewer0, 2020-10-13:
Just include a nozzle heating to 160°C (M109 S160 ; PLA) before homing in the start gcode, that will soften up the filament so that won't affect the homing. Or clean the nozzle after print jobs...

hamster65, 2020-10-13:
That is what I have now but it does not cover manual homing and it resulted in a crash yesterday.

qwewer0, 2020-10-13:
For manual homing you just need to manually heat up the nozzle before it.

Or if that is too much hassle for you, then you can make a `G28.gcode` file that has this in it:
```
M109 S160 ; PLA
G28
```

It will heat up the nozzle before homing.

hamster65, 2020-10-14:
Sure I could do that. But my FR was meant as a suggesting for putting in a protection against that issue.

qwewer0, 2020-10-14:
Marlin already has all the features to do just that.
So you either have to clean your nozzle after each print or use the "G28.gcode".

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: [https://discord.gg/n5NJ59y](https://discord.gg/n5NJ59y)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ManuelMcLure, 2020-10-14:
Many people (like myself) explicitly do _not_ want to heat up the nozzle before homing. My start gcode heats the bed, homes, moves the nozzle off the print bed and only then heats up the nozzle. Your proposed change would make it impossible for me to do that.
This is definitely something that is best handled through start gcode in the slicer.

qwewer0, 2020-10-14:
@ManuelMcLure If there would be a feature for this, then it wouldn't be default, so it wouldn't ruin your setup.
@hamster65 I'm pretty sure that even you would have more cases outside print jobs, where you wouldn't need to heat up the nozzle before homing and would curse that you enabled this feature, because then, you would have to switch off heating if you just want to quickly home or wait for it to complete.

> So you either have to clean your nozzle after each print or use the "G28.gcode".

G28.gcode:
> M109 S160 ; PLA
> G28

hamster65, 2020-10-14:
I can see your points. It seems my idea is specific to the issue I had with my printer and not not that useful to others. Closing.

closed 2020-10-14 by hamster65