MarlinFirmware/Marlin #19717: [BUG] M106 S255 (without P argument) does nothing

c3D-Dan, 2020-10-13:
M106 S255 does nothing
M106 P0 S255 works fine

M107 does nothing
M107 P0 works fine

Marlin2 bugfix

Cura guys say it's a marlin issue.  Apparently not having the P argument is documented to work in Marlin docs.

ellensp, 2020-10-13:
Please follow the Issue Template when filing a bug report. This includes attaching your configs:
 
```lisp
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

c3D-Dan, 2020-10-13:
This is a very simple bug.  Is it really required?

ellensp, 2020-10-13:
Always. Although it looks simple it may be motherboard specific or some other setting you have set that you haven't mentioned.

boelle, 2020-10-13:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-10-13:
> Cura guys say it's a marlin issue.

ask them to be very specific since they know that the error is with marlin

qwewer0, 2020-10-13:
M106 S255 - Works
M106 P0 S255 - Works
M107 - Works
M107 P0 - Works

@c3D-Dan There isn't any problem with Marlin on my setup, so it is most likely your configuration, or your board.

Latest Bugfix-2.0.x
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5374105/Configuration.zip)

sjasonsmith, 2020-10-28:
I am closing this since @qwewer0 has verified it works and the reporter will not provide configuration files.

closed 2020-10-28 by sjasonsmith