MarlinFirmware/Marlin #19719: Bugfix 2.0.x: last bugfix does not compile with digipot

Bergerac56, 2020-10-13:
When compiling with #define DIGIPOT_MCP4451 uncommented for a MKS-SBASE , I get an error:

Linking .pio\build\LPC1768\firmware.elf
c:/users/olivi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/feature/digipot/digipot_mcp4451.cpp.o:C:\Users\olivi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\feature\digipot/digipot.h:33: multiple definition of `digipot_i2c'; .pio/build/LPC1768/src/src/MarlinCore.cpp.o:C:\Users\olivi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\feature/digipot/digipot.h:33: first defined here
c:/users/olivi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/gcode/feature/digipot/M907-M910.cpp.o:C:\Users\olivi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\gcode\feature\digipot\../../../feature/digipot/digipot.h:33: multiple definition of `digipot_i2c'; .pio/build/LPC1768/src/src/MarlinCore.cpp.o:C:\Users\olivi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\feature/digipot/digipot.h:33: 
first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1768\firmware.elf] Error 1

I got the problem with my largly modified config files for a MKS SBASE (LPC1768).

So, to be sure that it was not related to some changes not transposed to my files, I tested with a plain vanilla version (and default config files). Just modified the motherboard type and activated digipot options

I join those files 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5372456/Marlin.zip)

thinkyhead, 2020-10-14:
Should now be patched!

closed 2020-10-14 by thinkyhead

Bergerac56, 2020-10-16:
Nice. It works
Thank a lot