MarlinFirmware/Marlin #19720: [BUG] (2.0.x "Error while detecting libraries included by ..." at "Detecting libraries" Arduino compilation phase)

Theoristos, 2020-10-13:
### Bug Description

I have a lot of "Error while detecting libraries included by [sketch temp path]/[filename.cpp]"
messages during the first phase ("Detecting libraries used") of compiling Mariln on Arduino platform when "verbose" output is set on.

### My Configurations

Arduino-1.8.13, 
bare bone Marlin 2.0.7 with default config's.

### Steps to Reproduce
Start Arduino. Open 'Marlin.ino'  . Do test compile with "verbose output at   Compiling" checkbox ON in Settings, 

A lot of messages like
`Detecting libraries used...`
`Error while detecting libraries included by C:\Users\[Username]\AppData\Local\Temp\arduino_build_[build No]\sketch\src\MarlinCore.cpp`
`Error while detecting libraries included by C:\Users\[Username]\AppData\Local\Temp\arduino_build_[build No]\sketch\src\gcode\calibrate\G28.cpp`
`...`
would be present.

#### Additional Information

Further investigation reveals that is due to erroneous macro' behavior due to unexpected values of some #define:

All such "bugged" files includes 'Configuration.h' or 'Configuration_adv.h', which in turn contain a set of "flag" empty #defines with a comment at the end of line like
`#define ARC_SUPPORT                 // Disable this feature to save ~3226 bytes`
`#define HOST_KEEPALIVE_FEATURE        // Disable this if your host doesn't like keepalive messages`

The problem it that at "Detecting libraries" phase Arduino calls gcc preprocessor with '-CC' ("keep the comments") option.
So, the 'ARC_SUPPORT' takes not empty value, but a '/* Disable this feature to save ~3226 bytes*/' one.

When onwards that flag is checked by 'ENABLED()' macro like
`... #if ENABLED(ARC_SUPPORT) ...`
that unexpected value fall through the guts of tons of nested macros, ending with preprocessing error with message like
`c:\users\[username]\appdata\local\temp\arduino_build_[build No]\sketch\src\core\macros.h:185:48: error: pasting "ENA_" and "/* Disable this feature to save ~3226 bytes*/" does not give a valid preprocessing token`
`#define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))`
(the macro wrongly identify this "empty" flag as single-argument variable but can't process it "value")

At main, sketch compiling phase -CC option is absent and preprocessing goes smoothly.

The severity of that bug is low, but potentially it will hinder third-part library identification if it #include-s stay lower than config #includes
Also, similar problems can occur not only with ENABLED() but similar macros like BOTH(), ANY(), ALL(), etc.

Possible ways to solve:
* Move all //comments after empty #define-s to other lines 
* Turn back from #if ENABLED(... to #ifdef varname
* Rework guts of checking macros to handle such problem

boelle, 2020-10-13:
> Arduino-1.8.13,

vscode+platform.io is the prefered tool

boelle, 2020-10-13:
reason for the above is that Arduino IDE which we have no control over what so ever have bugs that can prevent a compile

not saying its the cause here but its the mains reason for dev's use vscode+platform.io extension

boelle, 2020-10-13:
> Marlin 2.0.7

tip is to use [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) its updated more frequent, and 2.0.7 is just a frozen copy of it anyway

Vertabreak, 2020-10-13:
i support the removal of listing arduino as a viable method for compiling marlin there are several ongoing issues that prevent arduino from being able to compile marlin. these bugs cant be fixed and going forward i suspect the list of problem will grow. if the documentation no longer lists arduino as a option we will see far fewer of these reports.

boelle, 2020-10-13:
i would go to the extreme and say that this is not a bug with marlin, but i'm not the only one arround here

Theoristos, 2020-10-14:
It looks like everything has already been decided, but I'll try to put forward a pair of arguments for Arduino.

First. The compiling on Arduino is quite simple. Just download, run install, open ino, press "compile".
It's fine, when one have already downloaded both Visual Studio Code IDE and platform.io, correctly installed it, figured out how to run a compile, tune installation till it began to compile.  But I, for example have not have vscode installed and trained yet?

Second, even more significant. How to upload the result on arduino boards? The legal method was to plug a usb, run Arduino and pres the other one button. How to make this on vscode?  Are all these Reprap, Makerbase and other boards will become obsolete together with Arduino too?

At the end. Just curious, what is a real drawback to use #ifdef/#ifndef instead of #if ENABLED(...) macro?

sjasonsmith, 2020-10-14:
> First. The compiling on Arduino is quite simple. Just download, run install, open ino, press "compile".

Try that on Windows...

sjasonsmith, 2020-10-14:
OK, I see you are on Windows, but you must be failing out before you get to actually linking. Setting up VS Code / PlatformIO ends up far simpler than trying to trick Arduino to work properly with modern Marlin.

Theoristos, 2020-10-14:
> Try that on Windows...
Exactly. Win 7.

sjasonsmith, 2020-10-14:
Arduino on Windows only builds with one core. When you switch to PlatformIO, the build system intelligently decides which files to compile, and also spreads it across all your available cores. I just timed the build I happen to have open at 33 seconds.

Theoristos, 2020-10-14:
> Arduino on Windows only builds with one core. When you switch to PlatformIO, the build system intelligently decides which files to compile, and also spreads it across all your available cores. I just timed the build I happen to have open at 33 seconds.
Yes, it is real drawback of Arduino IDE. It tries to compile whole source files. Instead of 1.9.x,  2.0.x is highly inapplicable for such manner.

sjasonsmith, 2020-10-14:
For basic AVR-based boards? Just plug in the USB cable, and click the Upload button.

sjasonsmith, 2020-11-07:
I looked at this a bit. Eliminating trailing comments on every line in the configuration file would be a very large disruption to the style of the file. I would consider this an Arduino issue and not Marlin, since these trailing comments are perfectly acceptable in our compiled code.

Given the wide range of issues users encounter with the Arduino IDE, I don't think we would want to restructure our header file to eliminate what ultimately end up as cosmetic warnings during the Arduino compile.

closed 2020-11-07 by sjasonsmith