MarlinFirmware/Marlin #19722: [BUG] Marlin 2.0.7.1 or bugfix wont compile, RUMBA32 (Aus3d) with RepRap Discount Full Graphics Smart Controller

Celcius1, 2020-10-13:
I'm trying to compile the latest version of Marlin to fix the layer shifting issue but my firmware won't compile due to the error from the compiler output appears to be an issue regarding definition of pins etc.

n file included from Marlin/src/lcd/dogm/ultralcd_DOGM.cpp:42:
Marlin/src/lcd/dogm/ultralcd_DOGM.h:60:36: error: 'LCD_PINS_ENABLE' was not declared in this scope; did you mean 'DMA_PINC_ENABLE'?
   60 |     #define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS // AVR version ignores these pin settings
      |                                    ^~~~~~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
   78 | U8G_CLASS u8g(U8G_PARAM);
      |               ^~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_DOGM.h:60:53: error: 'LCD_PINS_RS' was not declared in this scope; did you mean 'LCD_PINS_D4'?
   60 |     #define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS // AVR version ignores these pin settings
      |                                                     ^~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
   78 | U8G_CLASS u8g(U8G_PARAM);
      |               ^~~~~~~~~
Compiling .pio/build/rumba32/src/src/lcd/menu/menu.cpp.o
*** [.pio/build/rumba32/src/src/lcd/dogm/ultralcd_DOGM.cpp.o] Error 1
In file included from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PinNames.h:4,
                 from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/pinmap.h:22,
                 from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PeripheralPins.h:34,
                 from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/analog.h:44,
                 from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/board.h:8,
                 from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/wiring.h:41,
                 from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:36,
                 from Marlin/src/lcd/dogm/../../inc/../HAL/./STM32/../shared/Marduino.h:36,
                 from Marlin/src/lcd/dogm/../../inc/../HAL/./STM32/HAL.h:28,
                 from Marlin/src/lcd/dogm/../../inc/../HAL/HAL.h:26,
                 from Marlin/src/lcd/dogm/../../inc/MarlinConfig.h:31,
                 from Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:30:
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp: In function 'void ST7920_SWSPI_SND_8BIT(uint8_t)':
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.h:31:25: error: 'LCD_PINS_ENABLE' was not declared in this scope; did you mean 'DMA_PINC_ENABLE'?
   31 | #define ST7920_DAT_PIN  LCD_PINS_ENABLE
      |                         ^~~~~~~~~~~~~~~
/Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PinNamesTypes.h:129:34: note: in definition of macro 'STM_PORT'
  129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
      |                                  ^
Marlin/src/lcd/dogm/../../inc/../HAL/./STM32/fastio.h:69:33: note: in expansion of macro '_WRITE'
   69 | #define WRITE(IO,V)             _WRITE(IO,V)
      |                                 ^~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:68:3: note: in expansion of macro 'WRITE'
   68 |   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
      |   ^~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:68:9: note: in expansion of macro 'ST7920_DAT_PIN'
   68 |   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
      |         ^~~~~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:76:3: note: in expansion of macro 'ST7920_SND_BIT'
   76 |   ST7920_SND_BIT; // 1
      |   ^~~~~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.h:31:25: error: 'LCD_PINS_ENABLE' was not declared in this scope; did you mean 'DMA_PINC_ENABLE'?
   31 | #define ST7920_DAT_PIN  LCD_PINS_ENABLE
      |                         ^~~~~~~~~~~~~~~
/Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PinNamesTypes.h:129:34: note: in definition of macro 'STM_PORT'
  129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
      |                                  ^
Marlin/src/lcd/dogm/../../inc/../HAL/./STM32/fastio.h:69:33: note: in expansion of macro '_WRITE'
   69 | #define WRITE(IO,V)             _WRITE(IO,V)
      |                                 ^~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:68:3: note: in expansion of macro 'WRITE'
   68 |   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
      |   ^~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:68:9: note: in expansion of macro 'ST7920_DAT_PIN'
   68 |   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
      |         ^~~~~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:77:3: note: in expansion of macro 'ST7920_SND_BIT'
   77 |   ST7920_SND_BIT; // 2
      |   ^~~~~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.h:31:25: error: 'LCD_PINS_ENABLE' was not declared in this scope; did you mean 'DMA_PINC_ENABLE'?
   31 | #define ST7920_DAT_PIN  LCD_PINS_ENABLE
      |                         ^~~~~~~~~~~~~~~
/Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PinNamesTypes.h:129:34: note: in definition of macro 'STM_PORT'
[... 33 lines cut ...]
   31 | #define ST7920_DAT_PIN  LCD_PINS_ENABLE
      |                         ^~~~~~~~~~~~~~~
/Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PinNamesTypes.h:129:34: note: in definition of macro 'STM_PORT'
  129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
      |                                  ^
Marlin/src/lcd/dogm/../../inc/../HAL/./STM32/fastio.h:69:33: note: in expansion of macro '_WRITE'
   69 | #define WRITE(IO,V)             _WRITE(IO,V)
      |                                 ^~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:68:3: note: in expansion of macro 'WRITE'
   68 |   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
      |   ^~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:68:9: note: in expansion of macro 'ST7920_DAT_PIN'
   68 |   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
      |         ^~~~~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:80:3: note: in expansion of macro 'ST7920_SND_BIT'
   80 |   ST7920_SND_BIT; // 5
      |   ^~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio/build/rumba32/src/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp.o] Error 1

I hope these compiler errors well help you guide me as to what is stopping the firmware from compiling.  As previous version had no issues, apart from the layer shifting issue

I've also included my config files for reference

[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/5372732/Configuration.Files.zip)

thisiskeithb, 2020-10-14:
You're telling Malrin you have 7 drivers (X + Y + Z0 + Z1 + Z2 + Z3 + E0) when there are only 6 spots available:

```cpp
//#define X_DRIVER_TYPE  A4988
//#define Y_DRIVER_TYPE  A4988
#define Z_DRIVER_TYPE  TMC2209
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE TMC2209
#define Z3_DRIVER_TYPE TMC2209
#define Z4_DRIVER_TYPE TMC2209
#define E0_DRIVER_TYPE TMC2209
```

and

```cpp
#define NUM_Z_STEPPER_DRIVERS 4   // (1-4) Z options change based on how many
```

The sanity check catches that:

```prolog
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:1544:8: error: #error "No E stepper plug left for Z4!"
 1544 |       #error "No E stepper plug left for Z4!"
```

Reducing `NUM_Z_STEPPER_DRIVERS` to 3 allows for a successful compile using VSCode + PlatformIO and the latest `bugfix-2.0.x`.

```prolog
======================================== [SUCCESS] Took 34.34 seconds ========================================

Environment    Status    Duration
-------------  --------  ------------
rumba32        SUCCESS   00:00:34.338
======================================== 1 succeeded in 00:00:34.338 ========================================
```

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Celcius1, 2020-10-14:
I should've Posted my pins file as well, cause I use Mechaduino's for the X and Y drivers and they are fed step and dir signals from the EXP3 header
[RUMBA Pins Files.zip](https://github.com/MarlinFirmware/Marlin/files/5374921/RUMBA.Pins.Files.zip)

And my marling 2.0.7.1 is still not compiling, before closing this on me give me a chance to respond!

Celcius1, 2020-10-14:
@thisiskeithb I've reopened this cause you jumped the gun you TWAT!

sjasonsmith, 2020-10-14:
> @thisiskeithb I've reopened this cause you jumped the gun you TWAT!

It is possible this was incorrectly closed, but calling names is not a good way to make anybody spend time helping.

@thisiskeithb spends a lot of time reviewing incoming requests, and has to quickly decide whether they are likely a misconfiguration or a bug. Occasionally mistakes are made, but it is no reason to get rude.

thisiskeithb, 2020-10-14:
> And my marlin 2.0.7.1 is still not compiling, before closing this on me give me a chance to respond!

You didn't provide all the required information to reproduce your problem.

> I've reopened this cause you jumped the gun you TWAT!

Calling people names is unacceptable. Before you respond again, please read the following from the Issue template:

> Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

ellensp, 2020-10-14:
HAS_SPI_LCD is now HAS_WIRED_LCD pins_RUMBA32_common.h

closed 2020-10-14 by ellensp

Celcius1, 2020-10-14:
@ellensp I just copied my pins file as I did not want to spend time going through making the changes as you can see I redefined multiple pins

ellensp, 2020-10-14:
you configs and pins file all work as expected with that one change on bugfix.

Celcius1, 2020-10-14:
@ellensp when did the define's get there names changed, as it has caught me out.