MarlinFirmware/Marlin #19725: [BUG] Power Loss Resume - Print Head does not move after resume

swanlm, 2020-10-13:
### Bug Description
Machine is a Creality Ender 5 with a BTT SKR Mini E3 V2 motherboard.
The firmware is configured for Power Loss Resume and PLR is enabled. A BigTreeTech Tech Mini UPS is installed.

After a power loss the printer goes through the resume steps but stops once the print head has moved to the XYZ position that it was at the time of power loss. This happens on two identical machines running the same hardware and firmware.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[MyConfigurations.zip](https://github.com/MarlinFirmware/Marlin/files/5373503/MyConfigurations.zip)

### Steps to Reproduce

1. With the print in progress, remove power from the printer.
2. As expected the print head raises 3mm off the part.
3. Upon restoring power, the LCD offers the option to resume the print. I select this option.
4. The print head homes on X and Y and the heaters start warming up.
5. Once the heaters are at temperature, the print head purges filament and then moves to the correct XY position and the correct Z height as it was when the print was interrupted. So far, all is as expected.
6. At this point the LCD shows printing but the print head does not move any further. This happens every time.

**Expected behavior:** [What you expect to happen]
The printer should continue the print from the position it was prior to power loss.
**Actual behavior:** [What actually happens]
The printer restores to the position it was prior to power loss but does not continue any further.
#### Additional Information
1. I have included the PLR file in my configurations.
2. I am using Bugfix 2.0.x which I just updated.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

swanlm, 2020-10-13:
This looks to be the same issue as: 
https://github.com/MarlinFirmware/Marlin/issues/19602#issue-714126527

thisiskeithb, 2020-10-13:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/19602

closed 2020-10-13 by thisiskeithb