MarlinFirmware/Marlin #19728: Bugfix 2.0.x.x Still Not compiling with RUMBA32 With Graphical LCD

Celcius1, 2020-10-14:
My Compiler Error Output:

In file included from Marlin/src/lcd/dogm/ultralcd_DOGM.cpp:42:
Marlin/src/lcd/dogm/ultralcd_DOGM.h:60:36: error: 'LCD_PINS_ENABLE' was not declared in this scope; did you mean 'DMA_PINC_ENABLE'?
   60 |     #define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS // AVR version ignores these pin settings
      |                                    ^~~~~~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
   78 | U8G_CLASS u8g(U8G_PARAM);
      |               ^~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_DOGM.h:60:53: error: 'LCD_PINS_RS' was not declared in this scope; did you mean 'LCD_PINS_D4'?
   60 |     #define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS // AVR version ignores these pin settings
      |                                                     ^~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
   78 | U8G_CLASS u8g(U8G_PARAM);
      |               ^~~~~~~~~
*** [.pio/build/rumba32/src/src/lcd/dogm/ultralcd_DOGM.cpp.o] Error 1
In file included from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PinNames.h:4,
                 from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/pinmap.h:22,
                 from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PeripheralPins.h:34,
                 from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/analog.h:44,
                 from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/board.h:8,
                 from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/wiring.h:41,
                 from /Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:36,
                 from Marlin/src/lcd/dogm/../../inc/../HAL/./STM32/../shared/Marduino.h:36,
                 from Marlin/src/lcd/dogm/../../inc/../HAL/./STM32/HAL.h:28,
                 from Marlin/src/lcd/dogm/../../inc/../HAL/HAL.h:26,
                 from Marlin/src/lcd/dogm/../../inc/MarlinConfig.h:31,
                 from Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:30:
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp: In function 'void ST7920_SWSPI_SND_8BIT(uint8_t)':
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.h:31:25: error: 'LCD_PINS_ENABLE' was not declared in this scope; did you mean 'DMA_PINC_ENABLE'?
   31 | #define ST7920_DAT_PIN  LCD_PINS_ENABLE
      |                         ^~~~~~~~~~~~~~~
/Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PinNamesTypes.h:129:34: note: in definition of macro 'STM_PORT'
  129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
      |                                  ^
Marlin/src/lcd/dogm/../../inc/../HAL/./STM32/fastio.h:69:33: note: in expansion of macro '_WRITE'
   69 | #define WRITE(IO,V)             _WRITE(IO,V)
      |                                 ^~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:68:3: note: in expansion of macro 'WRITE'
   68 |   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
      |   ^~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:68:9: note: in expansion of macro 'ST7920_DAT_PIN'
   68 |   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
      |         ^~~~~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:76:3: note: in expansion of macro 'ST7920_SND_BIT'
   76 |   ST7920_SND_BIT; // 1
      |   ^~~~~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.h:31:25: error: 'LCD_PINS_ENABLE' was not declared in this scope; did you mean 'DMA_PINC_ENABLE'?
   31 | #define ST7920_DAT_PIN  LCD_PINS_ENABLE
      |                         ^~~~~~~~~~~~~~~
/Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PinNamesTypes.h:129:34: note: in definition of macro 'STM_PORT'
  129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
      |                                  ^
Marlin/src/lcd/dogm/../../inc/../HAL/./STM32/fastio.h:69:33: note: in expansion of macro '_WRITE'
   69 | #define WRITE(IO,V)             _WRITE(IO,V)
      |                                 ^~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:68:3: note: in expansion of macro 'WRITE'
   68 |   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
      |   ^~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:68:9: note: in expansion of macro 'ST7920_DAT_PIN'
   68 |   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
      |         ^~~~~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:77:3: note: in expansion of macro 'ST7920_SND_BIT'
   77 |   ST7920_SND_BIT; // 2
      |   ^~~~~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.h:31:25: error: 'LCD_PINS_ENABLE' was not declared in this scope; did you mean 'DMA_PINC_ENABLE'?
   31 | #define ST7920_DAT_PIN  LCD_PINS_ENABLE
      |                         ^~~~~~~~~~~~~~~
/Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PinNamesTypes.h:129:34: note: in definition of macro 'STM_PORT'
  129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
[... 33 lines cut ...]
      |                         ^~~~~~~~~~~~~~~
/Users/celcius1-mac/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PinNamesTypes.h:129:34: note: in definition of macro 'STM_PORT'
  129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
      |                                  ^
Marlin/src/lcd/dogm/../../inc/../HAL/./STM32/fastio.h:69:33: note: in expansion of macro '_WRITE'
   69 | #define WRITE(IO,V)             _WRITE(IO,V)
      |                                 ^~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:68:3: note: in expansion of macro 'WRITE'
   68 |   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
      |   ^~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:68:9: note: in expansion of macro 'ST7920_DAT_PIN'
   68 |   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
      |         ^~~~~~~~~~~~~~
Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp:80:3: note: in expansion of macro 'ST7920_SND_BIT'
   80 |   ST7920_SND_BIT; // 5
      |   ^~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio/build/rumba32/src/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp.o] Error 1

I have downloaded the latest bug fix from the marlin website and this is the result I get.

I have attached my config and pins files as I use mechaduino's for my X and Y motors (they don't use onboard drivers only need step and dir signals), and 5 of the 6 stepper slots for my extruder and 4 Z motors as I run the 4 point z levelling.  As I have redefined my pins accordingly.

[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/5375406/Configuration.Files.zip)
[RUMBA Pins Files.zip](https://github.com/MarlinFirmware/Marlin/files/5375407/RUMBA.Pins.Files.zip)

thinkyhead, 2020-10-14:
Sorry, if you don't fill out the form we can't help.

closed 2020-10-14 by thinkyhead

thisiskeithb, 2020-10-14:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/19722