MarlinFirmware/Marlin #19729: [BUG] SAMD51 Compile Error with Marlin 2.0.7

Plebejer258, 2020-10-14:
### Bug Description

With Motherboard Adafruit_Grand_Central_M4 and 2 Serial Configured, the Compile failed ith the Error "Serialincrement was not decleared in this Scope" 

### My Configurations

My Configuration compiles fine with Marlin 2.0.6

### Steps to Reproduce

1. [First Step]
Select Motherboard ADAFRUIT_GRAND_CENTRAL_M4
2. [Second Step]
Select 2 Serial Ports First -1 second 1
3. [and so on...]

**Expected behavior:** [What you expect to happen]
Compile without error like Marlin 2.0.6
**Actual behavior:** [What actually happens]
Failed in HAL.h Line 38

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-10-14:
ADAFRUIT_GRAND_CENTRAL_M4 is not a valid motherboard

ellensp, 2020-10-14:
The only motherboard for SAMD51_grandcentral_m4 is BOARD_AGCM4_RAMPS_144 with #define Adafruit_Grand_Central_M4 

ie 
```
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_AGCM4_RAMPS_144
  #define Adafruit_Grand_Central_M4 
#endif

```
And this does generate the error

Plebejer258, 2020-10-14:
Sorry, i know.
I defined the BOARD_AGCM4_RAMPS_144 not ADAFRUIT_GRAND_CENTRAL_M4.
What i mean was i compile for the SAMD51 and this is the only one.
So the problem is the same with AGCM4_RAMPS_144.

GMagician, 2020-10-14:
That's really weird since INCREMENT macro is defined (gives an error if you try to redefine it). Can't figure out why it doesn't compile.

Plebejer258, 2020-10-15:
Hi,
this is the exact Error:
In file included from Marlin\src\HAL\shared\backtrace\../../../core/../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\shared\backtrace\../../../core/../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\backtrace\../../../core/serial.h:24,
                 from Marlin\src\HAL\shared\backtrace\backtrace.cpp:28:
Marlin\src\HAL\shared\backtrace\backtrace.cpp: In function 'bool UnwReportOut(void*, const UnwReport*)':
g:\avr\projekte\marlin2_207_agc\marlin-2.0.7\marlin\src\hal\samd51\hal.h:38:23: error: 'SerialINCREMENT' was not declared in this scope
I did redefine but this didn't change any behavior. 
My Configuration is very Basic, only the Board,, A988 Driver and the Reprap Graphic LCD. Nothing else.

ellensp, 2020-10-15:
a bisect shows issue is in commit 4b928b2da8ae26bd4a949debe470a914e23545ca
but that is just when the  INCREMENT macro started being used..  
Doesn't explain why INCREMENT macro isn't working.

ellensp, 2020-10-15:
tried latest toolchain-gccarmnoneeabi, made no difference.
platform_packages = toolchain-gccarmnoneeabi @ 1.90301.200702

GMagician, 2020-10-15:
I think it's a compiler error. Even redefining INCREMENT as 1 (`#define INCREMENT(X) 1`)  macro is not resolved correctly and generates an error

sjasonsmith, 2020-10-16:
This just needs an extra macro so that the increment is resolved before it is appended to `Serial`. I'll post a PR shortly.

sjasonsmith, 2020-10-28:
Closing this since a fix was merged and the issue has been silent.

closed 2020-10-28 by sjasonsmith