MarlinFirmware/Marlin #19732: ASSISTED_TRAMMING did not work.

Talha909, 2020-10-14:
HI, 

I have downloaded the new bugfix and enable the ASSISTED_TRAMMING. When I gave the command G35 then my nozzle goes into  middle and probe perfectly . After probing in the middle it gave me error |G35 failed at point 0 (Front-Left) X20.00 Y20.00,
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5376344/Marlin.zip)

closed 2020-10-14 by Talha909

hamster65, 2020-11-07:
Same for me with current bugfix. Did you solve it? After G35, it homes X,Y, Z and probes in the middle of the bed, but it does not move to the first tramming position. Output:

Recv: X:100.00 Y:142.00 Z:6.67 E:0.00 Count A:38720 B:-6720 Z:16008
Recv: G35 failed at point 0 (Front-Left) X10.00 Y20.00
Recv: G35 aborted.

After that, it probes again at the same position.

The LCD tramming assistant homes XYZ and does nothing more, no matter what I select or click.

Speaka, 2020-11-07:
In the initial problem was defined:
`#define Y_MIN_POS 0`
`#define NOZZLE_TO_PROBE_OFFSET { 0, 45, -0.420 }`
`#define TRAMMING_POINT_XY { {  20, 20 }, { 280,  20 }, { 280, 280 }, { 20, 280 } }`

So if you try to Probe the first point: 20/20 your nozzle will 45mm left of this point -> Y=-25. But this is in conflict with ymin = 0.
This is why it fails. Maybe similiar for @hamster65 

Solution: Move the probing points more inwards or increase the x/y min/max (**if physically acceptable**)

thisiskeithb, 2020-11-07:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

hamster65, 2020-11-07:
In that case Marlin should output an error message on the terminal to make this mistake obvious to the user.

My config is:
#define X_BED_SIZE 200
#define Y_BED_SIZE 213

#define X_MIN_POS -68
#define Y_MIN_POS -53

#define X_MAX_POS X_BED_SIZE+15
#define Y_MAX_POS Y_BED_SIZE+65

#define TRAMMING_POINT_XY { {  10, 20 }, { 190,  20 }, { 190, 190 }, { 10, 190 } }

#define NOZZLE_TO_PROBE_OFFSET { 0, -36, -1.67 }

So if I am not mistaken, that should work. Maybe a problem with negative coordinates?

hamster65, 2020-11-07:
I verified that nozzle _and_ probe can reach the tramming position. In fact, there is lot's of room around and I can use G1 to move to this positions.

I set  TRAMMING_POINT_XY to { {  20, 30 }, { 180,  30 }, { 180, 180 }, { 20, 180 } } now and enabled debugging. File attached.
[debug_leveling.txt](https://github.com/MarlinFirmware/Marlin/files/5505443/debug_leveling.txt)

What I see is:
- Homes XYZ
- Outputs "position not reachable", but it is....
- Probes in the middle of the bed (is this an error?)

hamster65, 2020-11-27:
I finally figured it out: It also takes #define PROBING_MARGIN into account!

This should be in the documentation. Something like "Make sure tramming points are within reachable position and probing margin.".

jonferreira, 2020-12-26:
@hamster65 mind expanding your answer a bit further?

hamster65, 2020-12-26:
All tramming points have to be within the probing area. It is not enough that they are within x and y min and max, You also have to subtract PROBING_MARGIN, and probably MESH_INSET as well. And of course shift all coordinates according to probe offset 

There is work in progress on sanity checks for that to give an error when compiling if the coordinates do not work.

tpruvot, 2020-12-26:
i think we should consider the system used for LEVEL_BED_CORNERS, basically a distance from bed borders