MarlinFirmware/Marlin #19736: [FR] Pin definition support for Closed Loop Stepper drivers i.e. Mechaduino, BTT S42B etc.

Celcius1, 2020-10-14:
In the driver type section is it possible to add an option to select a define method as External or Mechaduino as with most onboard drivers they use 4 control signals STEP, DIR, ENABLE and CS

As external closed loop drivers only really need two pins defined, STEP and DIR on most boards, RAMPS may need an enable.  but if I redefine these pins to other MCU pins and don't redefine the unneeded pins the sanity check throws an error.  is it possible to have a define option as either EXTERNAL, EXTERNAL_E

Where EXTERNAL is telling the sanity check to look for only STEP and DIR pins that have been defined in the pins file
Where EXTERNAL_E is telling the sanity check to look for STEP, DIR and ENABLE

Cause with this option it will allow me to utilise pins that are literally doing nothing but keeping the marlin sanity check happy.  As I'm looking to add more motors in the future.

But in general the feature would be good for those of us that use Closed Loop Drivers, to only use the pins that are needed, freeing the pins up for other uses

ellensp, 2020-10-14:
look in Configuration_adv.h

```
// Employ an external closed loop controller. Override pins here if needed.
//#define EXTERNAL_CLOSED_LOOP_CONTROLLER
#if ENABLED(EXTERNAL_CLOSED_LOOP_CONTROLLER)
  //#define CLOSED_LOOP_ENABLE_PIN        -1
  //#define CLOSED_LOOP_MOVE_COMPLETE_PIN -1
#endif
```

ellensp, 2020-10-14:
ok, that might be only half an answer...

Celcius1, 2020-10-14:
> ok, that might be only half an answer...

yeah it is more or less half an answer, I still have to deal with the assignment of the CS pin

ellensp, 2020-10-14:
but if you add SOFTWARE_DRIVER_ENABLE  the checks should allow you to not enable pins defined

Celcius1, 2020-10-14:
> but if you add SOFTWARE_DRIVER_ENABLE the checks should allow you to not have enable pins defined

does not appear to be the case it is designed more for the SPI and UART based drivers, let me give it a quick try and see if it works

Celcius1, 2020-10-14:
> but if you add SOFTWARE_DRIVER_ENABLE the checks should allow you to not have enable pins defined

enabled the option and disabled the pins for x and y enable and cs by commenting them out and the compiler threw this error

In file included from Marlin/src/MarlinCore.cpp:316:
Marlin/src/MarlinCore.cpp: In function 'bool pin_is_protected(pin_t)':
Marlin/src/pins/sensitive_pins.h:59:40: error: 'X_ENABLE_PIN' was not declared in this scope; did you mean 'Z_ENABLE_PIN'?
   59 | #define _X_PINS X_STEP_PIN, X_DIR_PIN, X_ENABLE_PIN, _X_MIN _X_MAX _X_MS1 _X_MS2 _X_MS3 _X_CS
      |                                        ^~~~~~~~~~~~
Marlin/src/pins/sensitive_pins.h:684:3: note: in expansion of macro '_X_PINS'
  684 |   _X_PINS _Y_PINS _Z_PINS _X2_PINS _Y2_PINS _Z2_PINS _Z3_PINS _Z4_PINS _Z_PROBE \
      |   ^~~~~~~
Marlin/src/MarlinCore.cpp:322:49: note: in expansion of macro 'SENSITIVE_PINS'
  322 |   static const pin_t sensitive_pins[] PROGMEM = SENSITIVE_PINS;
      |                                                 ^~~~~~~~~~~~~~
Marlin/src/pins/sensitive_pins.h:92:40: error: 'Y_ENABLE_PIN' was not declared in this scope; did you mean 'Z_ENABLE_PIN'?
   92 | #define _Y_PINS Y_STEP_PIN, Y_DIR_PIN, Y_ENABLE_PIN, _Y_MIN _Y_MAX _Y_MS1 _Y_MS2 _Y_MS3 _Y_CS
      |                                        ^~~~~~~~~~~~
Marlin/src/pins/sensitive_pins.h:684:11: note: in expansion of macro '_Y_PINS'
  684 |   _X_PINS _Y_PINS _Z_PINS _X2_PINS _Y2_PINS _Z2_PINS _Z3_PINS _Z4_PINS _Z_PROBE \
      |           ^~~~~~~
Marlin/src/MarlinCore.cpp:322:49: note: in expansion of macro 'SENSITIVE_PINS'
  322 |   static const pin_t sensitive_pins[] PROGMEM = SENSITIVE_PINS;
      |                                                 ^~~~~~~~~~~~~~
*** [.pio/build/rumba32/src/src/MarlinCore.cpp.o] Error 1

Celcius1, 2020-10-14:
Giving it more thought I would say if you select EXTERNAL option, you then go to the configuration_adv to specify the custom pins for the drivers specified and to use the ports for other drivers like In my case for extra Z drivers, have the define statements reassign them to z drivers etc.  I could attempt to modify the code myself and publish my attempts here for scrutiny