MarlinFirmware/Marlin #19737: [BUG] Part cooling fan starts at 100% when selecting "Preheat ABS Bed" or "Preheat PLA Bed"

lokster, 2020-10-14:
### Bug Description

I'm running Marlin-2.0.7 on RAMPS 1.6. Whenever I select "Temperature -> Preheat ABS -> Preheat ABS Bed" my part cooling fan starts running at full speed even though the Fan speed is set to 0% in the Preheat ABS/PLA Conf. 
This happens only when selecting Preheat ABS Bed (or Preheat PLA Bed). Does not happen when selecting "Temperature -> Preheat ABS -> Preheat ABS" or "Temperature -> Preheat ABS -> Preheat ABS End".

### My Configurations

[My Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5378874/My.Configuration.zip)

ellensp, 2020-10-14:
This setting is stored in eeprom, make sure you reset eeprom contents to match your firmware settings. M502 then M500

lokster, 2020-10-14:
It's the first thing I did after I flashed 2.0.7 and noticed the issue. Didn't fix it. To be sure, I just did it again, and the result is the same. Happens on 2 different printers with this configuration.

rhapsodyv, 2020-10-16:
Fixed here: #19773 
Can you test it?

PR #19773 (2020-10-17): Fix starting fan with bad material preset index
### Description

When pre heating only bed, the pre head code receives -1 to not start fan, but the code don't handle it, passing a random memory value to the set_fan_speed.

This fix that.

### Benefits

Fix #19737

### Related Issues

#19737
closed 2020-10-17 by thinkyhead