MarlinFirmware/Marlin #19738: [BUG] Z Probe wizzard restrarting printer.

th0m4sek, 2020-10-14:
Board MKS Robin Nano v1.2
I's using TFT_COLOR_UI and bltouch.
Wizard starts and printer gets restarted after homing XYZ.

boelle, 2020-10-14:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

th0m4sek, 2020-10-14:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5380747/Marlin.zip)
I'm on bugfix branch already.

rhapsodyv, 2020-10-15:
Color UI currently don't support Z Probe Wizzard.
```
Linking .pio/build/mks_robin_nano35/firmware.elf
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_probe_offset.cpp.o: In function `probe_offset_wizard_menu()':
/Users/victor/Development/Marlin/Marlin/src/lcd/menu/menu_probe_offset.cpp:93: warning: undefined reference to `lcd_moveto(unsigned char, unsigned char)'
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_probe_offset.cpp.o: In function `lcd_put_u8str(char const*)':
/Users/victor/Development/Marlin/Marlin/src/lcd/menu/../lcdprint.h:161: warning: undefined reference to `lcd_put_u8str_max(char const*, unsigned short)'
```

rhapsodyv, 2020-10-15:
Fixed here: #19742

PR #19742 (2020-10-15): Probe offset wizard for color ui
### Description

Fix probe offset wizard menu for color ui

### Benefits

Fix #19738

### Related Issues

#19738
closed 2020-10-18 by thisiskeithb