MarlinFirmware/Marlin #19741: Fan burst turn when sdcard.init is executed

creatron, 2020-10-14:
This is an 'old' bug, but persist in the latest version 2 bugfix, and has been submitted before, but is has been closed without being fixed. For a printer this is no problem at all, but is you use an Ender 3 with Creality Laser connected to the Fan, it is a huge problem. The Ender (Melzi) does not have a SD card detect, thus one has to 'Attach Media' (sd.initialse) the SD card, then it fires..  Thus every time you want to 'print' from the SD card the laser produce a burst  of energy..  I could not use the laser option  vailable in Marlin because the 1284 does not have a PWM pin available. This is the standard way Creality 'add' a laser on the printer. 

Reproduce, stock standard Ender3 config, or use my config files

[Marlinconfig.zip](https://github.com/MarlinFirmware/Marlin/files/5381281/Marlinconfig.zip)

.   Insert SD card, select 'Attach Media'.. and whala ,, the fan/led spin/light up. I added a LED on the fan output to enable the 'visibility' instead of a fan, as it is only a few honderd ms.  Indeed I wanted to test it with LED before connecting the laser. 

Thanks

sjasonsmith, 2020-11-07:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

The configs you provided are not for Marlin from this repo. We can only try to reproduce with configuration files which are compatible with default Marlin.

boelle, 2020-11-22:
closing since we dont support forks, 

we can reopen when configs from marlin are provided

closed 2020-11-22 by boelle

maassen, 2021-01-04:
Hello and happy new year,
I've the same problem with my Anet A8, and Marlin 2.0.7.2 with example configuration.
Maybe you can reopen this issue?
At startup and each sd card init, the cooling fan is activated.
I guess it's because the fan is connected to SS pin of the SPI Interface, and another pin is used as CS for the sd card.
Is it possible the deactivate SS pin during SPI transfer?
Regards, and thanks for this great software,
Mic

rhapsodyv, 2021-01-04:
Probably... but SS and CS should be, in theory, synonyms. What the code does, is try one pin, then the other. In your case, I think you just need to remove or set to -1 the wrong SS/CS (where the fan is connected)

maassen, 2021-01-04:
Maybe, but I think, the ATmega has a hardware SPI module, and this is used for accessing the sd card. And on the Anet board (and Melzi?) only MOSI, MISO, SCK are connected to sd card. CS of sd card is driven by an ordinary i/o pin. And CS or SS of the hardware module is used to drive the fan. It could be, that the CS pin is automatically driven, when the hardware SPI module is used. Maybe it would help, to set that pin as input during sd access.
Sorry I'm not an AVR expert.

rhapsodyv, 2021-01-04:
Yes, there're hard CS and soft CS, but none of them should be initiated if there's no device using them.

creatron, 2021-01-04:
Yes it is still a problem, I Just did an Ender 3 SKR E3 update and on the
32 bit board and seen the same problem. Problems can only be address when
one admit it is a problem.  I hoping to set some time aside to debug it, as
with the 32bit version and platformIO one can single step and place
breakpoints, as I do have a Segger ICE.

Kind Regards
Gerrie

-- 
Gerrie de Jager

(Quite people have the loudest minds.
Whether you think you can or you think you can't -- your'ed right.
Don't confuse having less with being less, having more with being more.
Who you are is far more important than what you have.
Ask busy people, others are 'busy')

Cell  072 198 1678

*www.creatron.co.za <http://www.creatron.co.za>*

This e-mail, its contents and any file attachments transmitted with it are
intended solely for the addressee(s) and may contain confidential
proprietary information. Access by any other party without the express
written permission of the sender is unauthorised. If you have received this
e-mail in error you may not copy, distribute or use the contents,
attachments or information in any way. Please destroy it and contact the
sender.

rhapsodyv, 2021-01-04:
> Problems can only be address when one admit it is a problem. 

You are asked to test it on bugfix or provide compatible configs, and just ignored us for two weeks, then it was closed. Not one said that there's no problem.

creatron, 2021-01-06:
The box standard config for Ender 3 can be used as available in the config
directory. I was just to busy during that time, it was year-end and I run a
business, thighs are just hectic. I used an Ender 3 box standard Melzi
mother board, as I replaced all my Enders with SKR 32Bit boards, and
installed this on an Eleksmaker A3 pro. as I wanted to 'burn' from sd card,
not using GRBL. I do however have Enders with Lasers but needed the 300mm,
that why I converted the Eleksmaker. Many people use an Laser on their
printers and use the FAN for Laser modulation, and that presents a problem,

Steps to produce:
Copy ender 3 config in marlin directory, enable SD card, program and press
reset. You can initialise the SD card as well from the menu, and then the
fan will also starts and stops.  I always have SD cards inserted in my
machines.

Kind Regards
Gerrie

-- 
Gerrie de Jager

(Quite people have the loudest minds.
Whether you think you can or you think you can't -- your'ed right.
Don't confuse having less with being less, having more with being more.
Who you are is far more important than what you have.
Ask busy people, others are 'busy')

Cell  072 198 1678

*www.creatron.co.za <http://www.creatron.co.za>*

This e-mail, its contents and any file attachments transmitted with it are
intended solely for the addressee(s) and may contain confidential
proprietary information. Access by any other party without the express
written permission of the sender is unauthorised. If you have received this
e-mail in error you may not copy, distribute or use the contents,
attachments or information in any way. Please destroy it and contact the
sender.

maassen, 2021-01-08:
Hello,
I looked a little bit into the code, and I found this in Marlin/src/HAL/AVR/HAL_SPI.cpp:

` void spiBegin() {`
`   OUT_WRITE(SS_PIN, HIGH);`
`   SET_OUTPUT(SCK_PIN);`
`   SET_INPUT(MISO_PIN);`
`   SET_OUTPUT(MOSI_PIN);`
`   ...`
Here SS is set high, even if it is not CS of SD Card. 
Maybe the `OUT_WRITE(SS_PIN, HIGH);` could be called only if SS is not output already.

ellensp, 2021-01-08:
This has already been address in bugfix https://github.com/MarlinFirmware/Marlin/pull/20606
```
  OUT_WRITE(SD_SS_PIN, HIGH);
  SET_OUTPUT(SCK_PIN);	  SET_OUTPUT(SD_SCK_PIN);
  SET_INPUT(MISO_PIN);	  SET_INPUT(SD_MISO_PIN);
  SET_OUTPUT(MOSI_PIN);	  SET_OUTPUT(SD_MOSI_PIN);
```

creatron, 2021-01-08:
Ok, thanks.

-- 
Gerrie de Jager

(Quite people have the loudest minds.
Whether you think you can or you think you can't -- your'ed right.
Don't confuse having less with being less, having more with being more.
Who you are is far more important than what you have.
Ask busy people, others are 'busy')

Cell  072 198 1678

*www.creatron.co.za <http://www.creatron.co.za>*

This e-mail, its contents and any file attachments transmitted with it are
intended solely for the addressee(s) and may contain confidential
proprietary information. Access by any other party without the express
written permission of the sender is unauthorised. If you have received this
e-mail in error you may not copy, distribute or use the contents,
attachments or information in any way. Please destroy it and contact the
sender.

maassen, 2021-01-08:
Okay, thank you. I'll try that.