MarlinFirmware/Marlin #19744: [BUG] extruder halts with TMC2208_STANDALONE and S_CURVE_ACCELERATION

RealDeuce, 2020-10-15:
### Bug Description

In addition to the well-reported Linear Advance extruder halt issue, I can reliably trigger the same thing with S_CURVE_ACCELERATION enabled.  With complex shapes, high speed, and a lot of relatively large retractions, the TMC2208_STANDALONE driven extruder will stop moving until it is disabled.  This seems to be yet another TMC2208 halt issue.

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5381930/Config.zip)

### Steps to Reproduce

With the attached config on a CR-10 v3 with stock board, print the attached gcode:
[SnakeBad2208.zip](https://github.com/MarlinFirmware/Marlin/files/5381934/SnakeBad2208.zip)

Within the first eight layers (usually within the first three), the extruder stepper will stop moving.  After the stepper is disabled, it will function normally again.

I tried various combinations of the min step, dir delay, and max steps values, and attempted "#define E0_DRIVER_TYPE A4988" to no avail.  After disabling S_CURVE_ACCELERATION (as in config file), the print succeeded.

A print without retractions printed successfully clearly suggesting something similar to what LA + TMC2208_STANDALONE does.  Slower speeds also print successfully, though I wouldn't trust it for a multi-day print.

**Expected behavior:** [What you expect to happen]

Extruder to continue rotating throughout the print.

**Actual behavior:** [What actually happens]

Extruder stops rotating.

#### Additional Information

There's a large number of bug reports surrounding TMC2208 and Linear Advance, but I couldn't find an existing one for S_CURVE_ACCELERATION.  This is likely either the same or similar issue as it seems to involve adjusting steps.  This is on an 8-bit board as well, so I suppose it's possible the CPU is just overloaded.

I'm not really equipped to check the chip temperature during a print, and don't really want to set the OTP bit to disable stealth as I don't have any extra boards and the drivers are not replaceable on the one I do have.

I also encountered the same issue when trying linear advance without s-curve accel, but that issue has been fairly well discussed.

RealDeuce, 2020-10-15:
And additional note, Octoprint was connected for all the prints, but one was attempted from the SD card with the same result (it actually failed closer to the beginning than the average).

Clarenceguoshuai, 2020-11-15:
1

rhapsodyv, 2020-12-01:
Is this a Marlin bug, or a limitation of TMC2208 for high speed and aggressive moving? I mean, is there any thing that we can fix at all?

boelle, 2020-12-01:
> is there any thing that we can fix at all

given that there has been no "usefull" info since 15 oct i think no, the reply 16 days ago is surely not usefull

boelle, 2020-12-01:
@teemuatlut do you have any ideas?

teemuatlut, 2020-12-01:
Considering this is with a standalone driver, we can't change the configuration nor can we read back any meaningful status data. This would also suggest it's likely an issue with S-curve where the steps would be generated too close to each other but I'm not too familiar with the feature and its coding. Preferably you'd set up an oscilloscope to trigger on a time where the pulses would be close and verify the behavior that way.

Thank you for the report but I'm afraid my only suggestion is don't use a feature that prevents you from using your printer. This likely won't be fixed until someone is invested enough to spend days/weeks/months on it.

closed 2021-01-09 by github-actions[bot]