MarlinFirmware/Marlin #19750: [BUG] Host computer seeing sdcard as corrupted

moham96, 2020-10-15:
### Bug Description

I'm connecting my board (skr 1.4) through a usb cable to my rapberry pi 3 expecting the raspberry pi to see the sdcard of the board as a usb storage device but the raspberry pi sees the sd card as corrupted appearantly 
```
[137293.395036] usb 1-1.3: new full-speed USB device number 22 using dwc_otg
[137293.528688] usb 1-1.3: New USB device found, idVendor=1d50, idProduct=6029, bcdDevice= 1.00
[137293.528710] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[137293.528725] usb 1-1.3: Product: Marlin USB Device
[137293.528737] usb 1-1.3: Manufacturer: marlinfw.org 
[137293.528750] usb 1-1.3: SerialNumber: 0BFF9019AF489C805D3A4A38F50020C7
[137293.529913] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
[137293.531112] usb-storage 1-1.3:1.2: USB Mass Storage device detected
[137293.531687] scsi host0: usb-storage 1-1.3:1.2
[137294.587339] scsi 0:0:0:0: Direct-Access     Marlin   SDCard 01        1.0  PQ: 0 ANSI: 0 CCS
[137294.592970] sd 0:0:0:0: [sda] 30547968 512-byte logical blocks: (15.6 GB/14.6 GiB)
[137294.594945] sd 0:0:0:0: [sda] Write Protect is off
[137294.594960] sd 0:0:0:0: [sda] Mode Sense: 00 00 00 00
[137294.602235] sd 0:0:0:0: [sda] Asking for cache data failed
[137294.602251] sd 0:0:0:0: [sda] Assuming drive cache: write through
[137301.429325]  sda: sda1
[137301.437823] sd 0:0:0:0: [sda] Attached SCSI removable disk
[137302.040988] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
[137302.041015] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x2 [current] 
[137302.041032] sd 0:0:0:0: [sda] tag#0 ASC=0x3a ASCQ=0x0 
[137302.041053] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 01 d2 1f 80 00 00 08 00
[137302.041071] blk_update_request: I/O error, dev sda, sector 30547840 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0[137302.042318] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
[137302.042337] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x2 [current] 
[137302.042353] sd 0:0:0:0: [sda] tag#0 ASC=0x3a ASCQ=0x3 
[137302.042371] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 01 d2 1f 80 00 00 08 00
[137302.042387] blk_update_request: I/O error, dev sda, sector 30547840 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[137302.042403] Buffer I/O error on dev sda, logical block 3818480, async page read

```
### My Configurations

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-10-17:
Have you tried formatting the SD card, using a different/known-good/name brand SD card, or using a different USB cable?

moham96, 2020-10-19:
> Have you tried formatting the SD card, using a different/known-good/name brand SD card, or using a different USB cable?

- I tried to format the SD card in fat32 and fat16 and they both showed the same problem.
- I used multiple SD cards and they all showed the same issue.
- I used two different cables and all showed the same problem.

**Note that** 
- the cards work just fine if i connect them to my pc or connect them to the raspberry pi using a card reader, Marlinfw also can read the contents of the card and print from it so this is not a problem in the cards.
- the UART interface works just fine and i can interact with marlin without connectivity issues so the cable is probably not the problem also.

Regards

shitcreek, 2020-10-21:
Did you 'release' media'?

moham96, 2020-10-21:
> Did you 'release' media'?

hmmm, I didn't know that i had to do that!
After releasing the sdcard the card can be mounted on the pi with no problem

Configuration_adv.h states the following:

> By default an onboard SD card reader may be **shared** as a USB mass-storage device

I think i remember (i might be just imagining) that i was able to mount the sdcard of my 8bit board (creality melzi) on the raspberry pi while it was `attached` inside marlin previously? so was that supprted previously or it's just my defective memory?

shitcreek, 2020-10-22:
Hey I'm glad that did the trick.

It depends on whether the board can emulate as a USB mass-storage device. I know for certain those 8bit Creality boards can't - I've just double checked the ones I've got here. Also the menu item shows as 'Change Media' but that doesn't seem to do anything.

Please close the thread if your issue has been resolved. Thanks.

PR #19932 (2020-10-29): Never Stop SPI on LPC
### Description

LPC MSC code start and configure the SPI 1 when the SD is release (to be shared with the host). After that initialisation, it relies that the SPI 1 will be always enabled and with the right configuration.

It's work fine with boards that keep SPI 1 only for SD.

But, some boards, like MKS SGEN L V2, share SPI 1 for the SD and the LCD/TFT. This makes SPI 1 changes constantly.

This results in random things, like not being able to mount SD, or even shows the SD as corrupted in the HOST.

This is just a temporary (but OK) fix, until we check if the LPC frame [... cut]
closed 2020-10-22 by moham96