MarlinFirmware/Marlin #19751: [FR] STM32 Nucleo-144 boards support based on STMicro STM32H7 Cortex-M7 MCU

Gamester17, 2020-10-15:
### Description

Requesting support for STMicroelectronics STM32 Nucleo-144 boards support based on STMicro STM32H7 Cortex-M7 MCU.

https://www.st.com/resource/en/data_brief/nucleo-h743zi.pdf

[image]

STM32H7 is 32-Bit microcontroller series based on ARM Cortex-M7 that has most variants clocked at 480MHz or 550MHz per MCU core, plus practically all variants also include embedded flash storage on-chip as well as featuring Ethernet PHY interfaces.

There are many STM32 Nucleo-144 development boards and evaluation boards model variants available from STMicroelectronics today, however, I suggest looking at the NUCLEO-H745ZI development board as the reference model of STM32 Nucleo-144 model as it features an STMicro STM32H745 (STM32H745ZIT6U) SoC which is a dual-core variant which has one core running at 480MHz and the other core running at 240MHz, 1MB RAM, 2MB flash memory and Ethernet PHY interface so that model covers most variations.

https://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-mpu-eval-tools/stm32-mcu-mpu-eval-tools/stm32-nucleo-boards/nucleo-h743zi.html

https://www.st.com/en/microcontrollers-microprocessors/stm32h745zi.html

This STMicro STM32 Nucleo-144 development board with manufacturer part number NUCLEO-H745ZI-Q samples goes for $29

https://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-mpu-eval-tools/stm32-mcu-mpu-eval-tools/stm32-nucleo-boards/nucleo-h743zi.html#sample-buy

Features and board specifications copied from https://www.st.com/resource/en/data_brief/nucleo-h743zi.pdf

* Common features
  * STM32 microcontroller in LQFP144 package
  * 3 user LEDs
  * 2 user and reset push-buttons
  * 32.768 kHz crystal oscillator
  * Board connectors:
    * SWD
    * ST Zio expansion connector including ARDUINO Uno V3
    * ST morpho expansion connector
  * Flexible power-supply options: ST-LINK, USB VBUS or external sources
  * On-board ST-LINK debugger/programmer with USB re-enumeration capability: mass storage, Virtual COM port, and debug port
  * Comprehensive free software libraries and examples available with the STM32Cube MCU Package
  * Support of a wide choice of Integrated Development Environments (IDEs) including IAR, Keil, and STM32CubeIDE

* Board-specific features
  * External or internal SMPS to generate Vcore logic supply
  *  Ethernet compliant with IEEE-802.3-2002
  * USB OTG full speed or device only
  * Board connectors:
    * USB with Micro-AB or USB Type-C
    * Ethernet RJ45
  * Arm Mbed Enabled compliant

Documentation, CAD resources, and software tools on their product page.

https://www.st.com/en/microcontrollers-microprocessors/stm32h7-series.html

https://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-mpu-eval-tools/stm32-mcu-mpu-eval-tools/stm32-nucleo-boards/nucleo-h743zi.html#documentation

#### Additional Information

Alternative models also with STM32H7 SoC are NUCLEO-H745ZI-Q, NUCLEO-H755ZI-Q, NUCLEO-H7A3ZI-Q, NUCLEO-L552ZE-Q.

The official STM32H735G-DK development kit for OEM developers costs $87 and comes with a 4.3-inch touchscreen display:

https://www.st.com/en/evaluation-tools/stm32h735g-dk.html?ecmp=tt17662_gl_link_sep2020#sample-buy

More information about the specifications in this article CNXSoft website:

https://www.cnx-software.com/2020/09/30/new-stm32h7-cortex-m7-mcus-clock-at-550-mhz-feature-octal-spi-flash-and-ethernet-interfaces/

https://www.cnx-software.com/2018/09/07/stmicro-stm32f7x0-h7x0-value-line-cortex-m7-mcu/

Disclaimer: I'm just an enthusiast and, of course, have no affiliations with STMicroelectronics or anyone who would profit from this. I just generally hope that we will see support for more boards based on powerful ARM Cortex-M7 cores in Marlin Firmware.

thisiskeithb, 2020-10-15:
It's a different variant, but support for the NUCLEO-F767ZI was just merged: https://github.com/MarlinFirmware/Marlin/pull/19373

Gamester17, 2020-10-15:
> It's a different variant, but support for the NUCLEO-F767ZI was just merged: #19373

Cool! Though that is not one of the newer STM32 Nucleo-144 boards which are all based on STMicro's new STM32H7 SoC series.

NUCLEO-F767ZI has a STM32F767ZIT6U SoC which has a STM32F767 (STM32F7x7 series) MCU with Cortex-M7 @ 216MHz.

https://www.st.com/en/microcontrollers-microprocessors/stm32f7x7.html

@sjasonsmith have you looked at the newer STM32 "Nucleo-144" board series with the new STM32H7 series SoCs by STMicro?

sjasonsmith, 2020-10-15:
I have a NUCLEO with an H7, but haven’t tried to add it yet. I was waiting for the F7 board to merge first to make sure I wasn’t wasting my time.

Gamester17, 2020-10-16:
Hopefully people in the community will make Nucleo adapter designs for RAMPS shields converting to Arduino Mega 2560 pinout.

As I understand it, STMicro made these Nucleo boards have same pinout layout design the Arduino Uno Rev3?

If and when such Nucleo adapters for RAMPS are available normal users should, in theory, be able to get started fairly quickly with it. 

Example @kursatu is building a Mega2560 converter adapter for Teensy 4.1 -> https://github.com/kursatu/Teensy4.1-ramps-shield

sjasonsmith, 2020-10-16:
The board I have is a STM32H743ZI, which is a single-code 480 MHz MCU.

Gamester17, 2020-10-16:
> The board I have is a STM32H743ZI, which is a single-code 480 MHz MCU.

That has same physical format of the board though, rights? So has same physical measurements and pinout/pins layout design?

sjasonsmith, 2020-10-16:
> That has same physical format of the board though, rights? So has same physical measurements and pinout/pins layout design?

Yes, although I haven't looked at them to see how closesly they resemble one another. I don't know whether all pins will be compatible between boards, or only the portions designed to receive official ST Nucleo shields.

Gamester17, 2020-10-16:
> Yes, although I haven't looked at them to see how closesly they resemble one another. I don't know whether all pins will be compatible between boards, or only the portions designed to receive [...]

I understand it as all "Nucleo-144" board headers should be fully pin-compatible with each other, "Nucleo-64" board headers should be fully pin-compatible with each other, and all "Nucleo-32" board headers should be fully pin-compatible with each other, but "Nucleo-144" and "Nucleo-64" headers are not fully pin-compatible with each other, instead they are only partially compatible. 

https://www.st.com/resource/en/user_manual/dm00244518-stm32-nucleo144-boards-mb1137-stmicroelectronics.pdf

https://www.st.com/resource/en/user_manual/dm00105823-stm32-nucleo-64-boards-mb1136-stmicroelectronics.pdf

That should mean that all shields made for "Nucleo-64" boards should be fully compatible with "Nucleo-144" board, but boards that are made for "Nucleo-144" and utilize all its extra pins will obviously not be fully compatible with "Nucleo-64" boards.

Here is an ST community discussion clarifying making shields for Nucleo boards and portability of shields between boards:

https://community.st.com/s/question/0D50X00009XkX2V/migration-from-nucleo64-board-to-nucleo144

The "ST Zio connector" on Nucleo-64 as well as Nucleo-144 boards both include Arduino Uno Rev3 and ST Morpho extensions:

- Arduino Uno Rev3 / Arduino Uno V3 connectivity.

- ST Morpho extension pin headers for full access to all STM32 I/Os (and only here is where Nucleo-144 differ from Nucleo-64).

Nucleo-144 with STM32F7 (STM32 F7) and STM32H7 (STM32 H7) also looks to be fully pin-compatible with each other.

https://www.st.com/en/evaluation-tools/nucleo-h745zi-q.html

https://www.st.com/en/evaluation-tools/nucleo-f767zi.html

* _"The ST Zio connector, which extends the ARDUINO Uno V3 connectivity, and the ST morpho headers provide an easy means of expanding the functionality of the Nucleo open development platform with a wide choice of specialized shields."_

* _"ST Zio connector including Arduino Uno V3, ST morpho (board connectors)"_

* "_Nucleo-64 include Arduino Uno rev3 & ST morpho connectors_"

* _"Nucleo-32 include Arduino Nano connectors"_

https://www.st.com/en/evaluation-tools/stm32-nucleo-boards.html

Gamester17, 2020-10-16:
Anyone know of more Nucleo-144 (or Nucleo-64) breakout-boards / shields / RAMPS BoB / ST expansion boards for 3D-printers?

According to @ghent360 it sounds as if Nucleo-64 boards with their 64-pins does not have enough pins for advanced 3D-printers:

https://www.youtube.com/watch?v=WxM7jmwh20s

@ghent360 has this 3D-printer controller board open-source hardware design for Nucleo-64 and a modified board with 100-pins:

https://github.com/ghent360/PrntrBoard

https://blog.pcbxprt.com/index.php/2018/05/11/prntr-board-v1/

Gamester17, 2020-10-16:
FYI, apparently, @hasenbanck looks to have developed an STM32H7 based 3D printer controller board he calls "Heteromycin":

https://github.com/hasenbanck/heteromycin

Though it seems like he has abandoned that project? And same with his STM32F7 3D printer controller board called "RemRam"?

https://github.com/hasenbanck/remram

Both projects are however open source hardware licensed.

Gamester17, 2020-10-16:
@thinkyhead missed you had made and closed PR https://github.com/MarlinFirmware/Marlin/pull/16021 last year for Nucleo STM32H7 support based on https://github.com/MarlinFirmware/Marlin/pull/13657 by @dekra54

@dekra54 you mentioned in https://github.com/MarlinFirmware/Marlin/pull/13657 that you are a hardware guy and were then working on something for it, any news on that?

Hedda, 2020-10-22:
The new BigTreeTech SKR SE-BX V1.0 board inside the upcoming BIQU BX 3D-printer is also based on an STM32H7 series MCU

https://www.kickstarter.com/projects/1440007055/biqu-bx-worlds-lightest-direct-extruder-fdm-3d-printer

BIQU specifically state STM32H743IIT6 as the chip used in their video running the 32-bit ARM Cortex-M7 core at 400MHz

https://www.youtube.com/watch?v=Yu3xfzXXBYA

The "400 MHz" was maybe confusing people since STMicro list the Cortex-M7 32-bit RISC core operating at "up to 480 MHz"

https://www.st.com/content/st_com/en/products/microcontrollers-microprocessors/stm32-32-bit-arm-cortex-mcus/stm32-high-performance-mcus/stm32h7-series/stm32h743-753/stm32h743ii.html

Presumably that BTT SKR SE-BX V1.0 board or a variant of it will also be sold as stand-alone upgrade boards from BIQU.

ghent360, 2020-10-25:
There is STM32H750VB, chip which is relatively affordable. I made a version of my PrntrBoardV2 with that chip - it is very similar pinout to F407. However it has a downside - flash is only 128K, compared with 512k on the F407VE. One can use SPI flash to store the firmware and load it to ram or use sd-card, but it is more hustle.

It is also not clear what benefit the high clock rate does offer for Marlin, it runs quite happily on sub 100MHz MCUs.

ghent360, 2020-10-25:
After I finished my Nucleo-64 based board, I briefly considered using Nucleo-144. I decided against it because these nucleo boards are gigantic. You will loose quite a bit of board real estate, just in the huge headers needed to connect the two boards.

Hedda, 2021-01-19:
FYI, STM32H7XX M devboards by DevEBox based on STM32H750VBT6 & STM32H743VIT6 is now available for $12-$18 from China

https://stm32-base.org/boards/STM32H750VBT6-STM32H7XX-M.html

https://stm32-base.org/boards/STM32H743VIT6-STM32H7XX-M.html

https://github.com/elzoughby/STM32H7xx-Startup

https://www.aliexpress.com/item/4000300005466.html

https://www.banggood.com/STM32H750VBT6-or-STM32H743VIT6-STM32H7-Development-Board-STM32-System-Board-M7-Core-Board-TFT-Interface-with-USB-Cable-p-1661383.html

[image]

[image]

[image]

Simple 3D model of STM32H7 DevEBox board

https://grabcad.com/library/stm32f7-devebox-board-prototype-1

It appears to use the same board design as the older STM32F407VGT6 based STM32F7XX M development board by DevEBox

https://github.com/mcauser/MCUDEV_DEVEBOX_F407VGT6

Thus I would assume that the DevEBox STM32H7XX M devboard is probably pin-compatible with DevEBox STM32F7XX M.

This is just an FYI as I have no affiliation what so ever with any manufacturers or any resellers of this of any other boards.

dreamcat4, 2021-01-19:
Thanks for the information. Currently speaking for a more proven solution we are stopped at the f4 over on the klipper firmware. (not marlin). The open source team at annex engineering have released klipper configurations for the fysetc s6 controller. Which is the current best. I asked the author of klipper software if he thought there would be any meaningful improvement in performance going from the f4 to the h7 and the response was no, not really.

Still... I would suggest it would not hurt to email somebody at fsetc to ask them if they are already working to make a new controller board based on the h7. Since given the very low cost of this MCU it would not impact the bom cost very much. And maybe if their existing s6 has already been out for a while.

Fsetc are being recommended as having good silk screen and a well thought out design. And the existing annex users feel their boards are overall better than the equivalents from BIQU / BTT.

slowbro, 2021-06-20:
It looks like stm32duino has added variant files for the H7 chips: https://github.com/stm32duino/Arduino_Core_STM32/commit/2f7d6884e57566aa3672ee0049f9a1cbbd1c3d46

thisiskeithb, 2021-06-20:
> It looks like stm32duino has added variant files for the H7 chips: stm32duino/Arduino_Core_STM32@2f7d688

Marlin also supports the Biqu BX under the STM32 HAL which uses an STM32H743IIT6:

https://github.com/MarlinFirmware/Marlin/blob/a006752fc2fde7a1f13a57eea190e2959fa19094/ini/stm32f7.ini#L42-L65

ag88, 2021-11-25:
a peek at a near future, hope they works well ;)
https://github.com/stm32duino/Arduino_Core_STM32/pull/1559
https://github.com/stm32duino/Arduino_Core_STM32/pull/1552

thisiskeithb, 2023-07-19:
> Requesting HAL support for STMicroelectronics STM32 Nucleo-144 boards based on newer STMicro STM32H7 Cortex-M7 MCU.

As far as HAL support goes, Marlin supports *many* STM32H7-based boards now, including the 144-pin H7 variant in the Octopus Max EZ. Take a look at [`stm32h7.ini`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/ini/stm32h7.ini) for details:

- STM32H743IIT6
- STM32H743VIT6
- STM32H723VGT6
- STM32H723ZET6

> https://www.st.com/resource/en/data_brief/nucleo-h743zi.pdf

If a developer wants to add a specific Nucleo board, they can submit a PR.

closed 2023-07-19 by thisiskeithb