MarlinFirmware/Marlin #19757: [BUG] RE: #19731 digipot won't compile

cptnmidnight, 2020-10-15:
I have tried to download every way possible, zip file git selected bugfix and main. How do I get the patch from this fix?

`/home/~/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-e
abi/bin/ld: .pio/build/LPC1768/src/src/feature/digipot/digipot_mcp4451.cpp.o:/home/~/Public/Marlin/Marlin/src/f
eature/digipot/digipot.h:33: multiple definition of digipot_i2c; .pio/build/LPC1768/src/src/MarlinCore.cpp.o:/home/~/Public/Marlin/Marlin/src/feature/digipot/digipot.h:33: first defined here`

thisiskeithb, 2020-10-15:
Download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-15 by thisiskeithb

cptnmidnight, 2020-10-15:
I think it is a bug still. Tried bugfix zip and git clone https://github.com/sjasonsmith/Marlin.git <-- I think this is the source of the patch, correct?

Still won't compile with line 1021 of configuration_adv.h uncommented

//#define DIGIPOT_MCP4451 <-- this compiles fine
#define DIGIPOT_MCP4451 <-- this gives an error "multiple definition of digipot_i2c"

thisiskeithb, 2020-10-15:
Use Marlin from my link above as that's from this repo.

cptnmidnight, 2020-10-15:
First thanks for looking into this for me. It's the first time I have not been able to solve a compile issue on my own.

I have tried all downloads available from this repo. I even tried sjasonsmith repo.

steps taken
unzip
change platformio.ini to LPC1768
change configuration.h motherboard def to BOARD_MKS_SBASE
change configuration_adv.h uncomment DIGIPOT_MCP4451

I will move back to the forums and see if anyone there has any ideas.

sjasonsmith, 2020-10-15:
My change was merged into bugfix-2.0.x, but I haven’t tried it since.

cptnmidnight, 2020-10-16:
Finally figured it out. I can say I am not a fan of github. I miss the days when diffs where the main post and not buried under ten links.

https://github.com/MarlinFirmware/Marlin/pull/19731/commits/29356e390323ed4d405e520ad3d4d2b7efcff008

this link has the patch and it compiles now