MarlinFirmware/Marlin #19758: [BUG] (short description)

Foxies-CSTL, 2020-10-15:
### Bug Description
Uncomment #define PAUSE_BEFORE_DEPLOY_STOW and active UBL on configuration Marlin to secure the deploy/stow Zprobe. Make a AutoCalibrationDelta, a test Zprobe and to build a manual mesh.

The PR #19694 resolv the "touch" on the TFT to valid the deploy/stow of Z_probe but only for the DeltaCalibration(G33) or the probe test (M48). For a build a manual mesh (Cold Mesh), **screen TFT freeze after TFT touch.**

### My Configurations
**Hardware:**FLSUN QQS-PRO with MKS Robin_mini 2.0 or Hispeedv1 boards with Mks Robin V2.0 (320x240).
**Software:** Bugfix 2.0.7.2, env VSC/PIO robin_mini and robin_nano35 for hispeed (aka clone robin_miniv2)

Include a ZIP file containing my configuration for standard Board MKS_Robin_mini and the new FLSun Board HISPEED.

### Steps to Reproduce
By the TFT menu or a terminal:
When I want to build a cold mesh using the UBL Tools menu or launch a bed calibration procedure using a terminal after initialization (HOME) then the deployment message, I touch the screen to validate, my screen freezes and I have to do a reset. No problem with the screen in "TFT_Classic_UI" mode only in "TFT_Color_UI" mode.
[Board_Robin_mini.zip](https://github.com/MarlinFirmware/Marlin/files/5388338/Board_Robin_mini.zip)
[Board_Hispeed.zip](https://github.com/MarlinFirmware/Marlin/files/5388339/Board_Hispeed.zip)

[image]

rhapsodyv, 2020-10-16:
Hi @Foxies-CSTL, thanks for your issue. I will work on that.

But I'm not familiar with UBL. I never used it. Can you post a video, showing all the steps until it get freeze?

Thanks!

rhapsodyv, 2020-10-16:
I think I found the issue :-).

Can you test it: #19771

Thanks

thisiskeithb, 2020-10-17:
@Foxies-CSTL Now that #19771 is merged, please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Foxies-CSTL, 2020-10-17:
Hi, 
I test with patch #19771 and the same issue. I put two videos. one with a reset then an auto delta calibration with the probe deployment / storage messages and another just after the Delta calibration with a probe test (M48) and a cold mesh request for UBL tools. This is where the screen freezes.

[1-Reset+DeltaCalibration.zip](https://github.com/MarlinFirmware/Marlin/files/5394898/1-Reset%2BDeltaCalibration.zip)
[2-TestProbe+UBL-Tools.zip](https://github.com/MarlinFirmware/Marlin/files/5394900/2-TestProbe%2BUBL-Tools.zip)

rhapsodyv, 2020-10-17:
That PR got changed in the way and may be broken. Can you try bugfix?

We found I few issues, but I really didn’t know where to look. 

Now with the videos could be easier. thanks! I will take a look now.

Foxies-CSTL, 2020-10-17:
Ok, I just dowload the last, I test it with succes!! But my printer reset alone after that. I have another problem now. I go to compare the config files.

rhapsodyv, 2020-10-17:
Ok, we get a little further. Sadly I don’t have a full setup similar of yours. But we can solve it!!

Your printer could be resetting because the watchdog. Try disable it to test.

If is the watchdog the problem, I need to know which is the screen/operation that you are doing when it reset. It could be missing a “watchdog refresh call”.

Foxies-CSTL, 2020-10-17:
Well, I checked my config and updated bugfix news, I made several tests by the menu and by Octoprint. Put back the options that I had removed (various menus, Neopixel, end of filament, UART mode on XYZ). Everything is operational with the Marlin Color user interface !!
Thank you for your help.

closed 2020-10-17 by Foxies-CSTL