MarlinFirmware/Marlin #19759: [BUG] UBL verification print

rq3, 2020-10-15:
After installing Marlin 2.0.7.2, the UBL function seems to work well, but the verification prints accelerate the print head at alarming speeds. The steppers ignore speed and acceleration limits defined in the configuration files.

ManuelMcLure, 2020-10-16:
I assume you're referring to the `G26` mesh print? There should be no way it's overriding the speeds, since it just feeds moves to the planner. It sends moves with extrusion (lines and circles) with 1/10th of the maximum X Y speed, and travel moves with 1/3 the defined feedrate. It does nothing special, so those should also be limited by the planner as well if by any chance they are past the limits.
Use `M503` to make sure that the values in your EEPROM actually match the values you have set in your configuration. Some slicers (Cura) will attempt to override the maximum speed and acceleration values with abnormal settings and those may have been left behind by an earlier print.

Vertabreak, 2020-10-16:
played with G26 for a few months a while back not once did i ever get a viable test print from it always just made a mess and it does have some very quick moves i would suggest never using it.

phazei, 2020-10-17:
I re-UBL'd my entire bed yesterday, tried the test print, the first part of the verticies it printed had really good adhesion, but after a very short while, a dozen circles or so, it was just spitting out in the air about 1mm away from the bed.  Today I merged in the latest marlin branch, noticed the z wizard isn't in the UBL menu (was hoping to use it) and am about to print the mesh as soon as it finished probing again.

closed 2020-11-24 by github-actions[bot]