MarlinFirmware/Marlin #19761: [FR] New definition pins for Delta QQS-Pro.

Foxies-CSTL, 2020-10-16:
### Description
FLSun delivers with the latest models of QQS-PRo, a clone of the Mks_Robin_min v2 board which allows drivers to be replaced. This is called HISPEEDv1. I created a new definition of pinouts to support different features.

### Feature Workflow
Features|Active|Value
  --------|------|-----
  Multi-Config Switch QQS-PRO|True|QQS_Config.h
  Choice of independent E driver from XYZ|Standard-TMC|Ready
  Module Wifi|True|Ready
  ZoffSet ED3v6|-|Ready|
  Extruder TITAN/BMG| Titan|Ready
  BabyStepping|True|Ready
  UI Type|True|Color Marlin|
  TFT Color Selection|True|Classic,Color
  Unified Bed Leveling|True|+53pts
  Filament sensor|True|Ready
  Recovery PowerFailed|True|-
  TMC 220x Soft Serial|UART|Ready
  TMC 2209 Hard Serial|UART|Ready
  Neopixel|Note|Ready

**Note:** For NeoPixel I am using a modified version 2.4.

For compilation under VSC/PIO, I use the modified environment of Robin_nano35 with the script "ms_robin_mini.py" for the binary naming and the script "add_nanolib.py "to reduce its size.
`[env:flsun_hispeed]
platform        = ${common_stm32f1.platform}
extends         = common_stm32f1
board           = genericSTM32F103VE
platform_packages = tool-stm32duino
extra_scripts   = ${common.extra_scripts}
  buildroot/share/PlatformIO/scripts/mks_robin_mini.py
  buildroot/share/PlatformIO/scripts/add_nanolib.py
build_flags     = ${common_stm32f1.build_flags}
  -DMCU_STM32F103VE -DSS_TIMER=4
lib_deps = ${common.lib_deps}
  SoftwareSerialM
  Adafruit NeoPixel=https://github.com/Foxies-CSTL/NeoPixel-Lib/archive/master.zip`

### Pack: Configs, Scripts, pins
[Pack_FLSun_QQS-Pro.zip](https://github.com/MarlinFirmware/Marlin/files/5388644/Pack_FLSun_QQS-Pro.zip)

#### Additional Information
My first stable release with Marlin Firmware 2.0.7 is on my [GitHub](https://github.com/Foxies-CSTL/Marlin_2.0.x/releases) . It contains the sources and different binaries to use.
_MB-FLSun_HiSpeed_
[image]
_MB-HiSpeed UART Serial mode_
[image]
_Function NeoPixel_
[image]
_Standard UI_
[image]

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Best regards,

sjasonsmith, 2020-10-16:
@Foxies-CSTL you will have much better success getting new board support added by opening a pull request yourself.

thisiskeithb, 2020-10-16:
> @Foxies-CSTL you will have much better success getting new board support added by opening a pull request yourself.

That and we won't be able to merge in the unified config. You'll need to split it up and contribute them to the Marlin Configurations repo as individual configs for easier maintainability.

Foxies-CSTL, 2020-10-16:
Hi, 
Thanks, I suspected that I was in the wrong section to declare this new definition of the board. I will try to do it by a PR but I have some problems in English with the syntax. 
Is there a French guy in the room? ;-) If I understand, I must make a request in the "Marlin_Configurations_repo" section for the "configurations" part and another in the Marlin_firmware section for the "pins definition" part.

Regards

thisiskeithb, 2020-10-16:
>  If I understand, I must make a request in the "Marlin_Configurations_repo" section for the "configurations" part and another in the Marlin_firmware section for the "pins definition" part.

Correct.

sjasonsmith, 2020-10-16:
I suggest focusing on adding the board support to the Marlin repo first. You don't need to worry about configurations until after the board is supported.

sjasonsmith, 2020-10-16:
We definitely do **not** want configurations before the board.

Foxies-CSTL, 2020-10-17:
Ok, thank's I will write a Pull Request for "my" board.
See you later ;-)

closed 2020-10-18 by Foxies-CSTL