MarlinFirmware/Marlin #19763: [BUG] Ender 5 Pro invert parameter not working

ArsenicBismuth, 2020-10-16:
This is a branch on the Ender 5 Pro 4.2.2/4.2.7 issue: https://github.com/MarlinFirmware/Marlin/issues/19506#issuecomment-709683722.

As stated in the title, the INVERT_X_DIR and INVERT_Y_DIR don't affect anything.

I was trying to make the endstops as the origin (0,0) rather than (MAX, MAX). So I changed these parameters:

```
#define INVERT_X_DIR true // was false
#define INVERT_Y_DIR true // was false

#define X_HOME_DIR -1 // was 1
#define Y_HOME_DIR -1 // was 1

#define USE_XMIN_PLUG // was the XMAX
#define USE_YMIN_PLUG // was the YMAX
```

But the X, Y invert stays the same, reverting back to false gave the same exact behavior. The firmware update does work as the home is indeed changed, just that it will move in the wrong direction and won't find the endstops.

A little bit of digging with similar problem: https://github.com/MarlinFirmware/Marlin/issues/10492. At least it indicates that there might be some advanced parameters I don't know, that prevent inverting. Please let me know if there's possibility of such parameters.

thisiskeithb, 2020-10-16:
The Ender-5 homes to X & Y max, not min, so the original settings are correct and not a Marlin bug.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-16 by thisiskeithb

ArsenicBismuth, 2020-10-16:
My Ender 5 homes to 0,0 tho out of the box, and I was trying to replicate that. But that's beside the point of the issue, in which Marlin doesn't allow me to invert the stepper direction (similar to that one Issue I stated at the end, except I don't know the parameter involved).

Tho yeah, maybe not a major issue. It's just a very strangely basic one.