MarlinFirmware/Marlin #19764: m280 servo command wierd behaviour

ekaggrat, 2020-10-16:
### Bug Description

i am using marlin 2.0 on a bigtreetech srk 1.3 board on a custom corexy. I am trying to use it for some simple pick and place operations with a servo mounted to the head. I just need the head to go to a origin location pick a block and then place it at a required destination. I wrote some gcode to test the movement . If i step the commands one by one in pronterface they all work fine . But if i load it as a file in pronterface and then run the gcode the servo opens and closes at the wrong time and after a few moves completely stops closing . I tried various combinations and get the same result. I am not sure where the error is occuring. 

### My Configurations
corexy , bigtreetech skr 1.3 a4988 16x 

### Steps to Reproduce

sample code i used :

G28 X0 Y0
G1 X5 Y5 F4000

M280 P0 S135
G1 Z12 F300
G1 X50 Y100 F4000
G1 Z0 F300
M280 P0 S45
G1 Z12 F300

G1 X5 Y5 F4000
G1 Z0 F300
M280 P0 S135
G1 Z12 F300
G1 X70 Y100 F4000
G1 Z0 F300
M280 P0 S45
G1 Z12 F300

G1 X5 Y5 F4000
G1 Z0 F300
M280 P0 S135
G1 Z12 F300
G1 X90 Y100 F4000
G1 Z0 F300
M280 P0 S45
G1 Z12 F300

G1 X5 Y5 F4000
G1 Z0 F300
M280 P0 S135
G1 Z12 F300
G1 X110 Y100 F4000
G1 Z0 F300
M280 P0 S45
G1 Z12 F300

boelle, 2020-10-16:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thisiskeithb, 2020-10-16:
You likely need to add some [`M400`](https://marlinfw.org/docs/gcode/M400.html)s in there to ensure moves are completed before continuing on.

ellensp, 2020-10-16:
Servos are not synced...  you defiantly need M400 before every M280, if you expect the servo action to occur after the move is completed

closed 2020-10-16 by thisiskeithb