MarlinFirmware/Marlin #19770: [FR] Movement Resolution of 0.001 in Z Probe Wizard

ils15, 2020-10-16:
### Description

When i adjust Z probe OFFSET, the best way to do a precise adjustment is combining babysteping with M851 in  #define BABYSTEP_ZPROBE_OFFSET          

But new Probe Offset Wizard only Show 0.025 Movement

### Feature Workflow

1. Select Probe Offset Wizard from Advanced Settings>Probe Offsets
2. Movement Menu shows Only these Options
a) Move Z 1mm
b) Move Z 0.1mm
c) Move Z 0.025mm

The best way is activate option Move Z 0.001mm when  #define BABYSTEP_ZPROBE_OFFSET  is active

##Benefit

More reliable and real world calibration settings

qwewer0, 2020-10-16:
> But new Probe Offset Wizard only Show 0.025 Movement

Because you have `SHORT_MANUAL_Z_MOVE` set to `0.025`.

thisiskeithb, 2020-10-16:
Reduce `SHORT_MANUAL_Z_MOVE` if you need finer adjustments.

closed 2020-10-16 by thisiskeithb