MarlinFirmware/Marlin #19772: [FR] Implement BLTOUCH_FORCE_5V_MODE

marcusletric, 2020-10-16:
### I couldn't find BLTOUCH_FORCE_5V_MODE anywhere in the source

On my Anet board I have to issue the force 5V command to make use of BLTouch / 3DTouch V3

```M280 P0 S140```

This is described in their documentation: https://5020dafe-17d8-4c4c-bf3b-914a8fdd5140.filesusr.com/ugd/f5a1c8_26f2661f0a4542eaacc1586eba8842e7.pdf

### Feature Workflow

1. when ```BLTOUCH_FORCE_5V_MODE``` is enabled, issue ```M280 P0 S140``` command on startup
#### Additional Information
I've got an Anet A6 printer and I use the LCD cable breakout on pin29 for the servo signal.

During my setup, I've changed the Marlin source and added this command on the setup phase. 
I've not done extensive tests yet, but this seems like it fixed my auto-home function and bed leveling. The probe now triggers properly.

FYI: I've used an old version of Marlin, but did not found this in the lastest version either.

GMagician, 2020-10-16:
Isn't it `BLTOUCH_SET_5V_MODE` in configuration_adv.h?

Vertabreak, 2020-10-16:
/**
   * Danger: Don't activate 5V mode unless attached to a 5V-tolerant controller!
   * V3.0 or 3.1: Set default mode to 5V mode at Marlin startup.
   * If disabled, OD mode is the hard-coded default on 3.0
   * On startup, Marlin will compare its eeprom to this value. If the selected mode
   * differs, a mode set eeprom write will be completed at initialization.
   * Use the option below to force an eeprom write to a V3.1 probe regardless.
   */
  //#define BLTOUCH_SET_5V_MODE

thisiskeithb, 2020-10-16:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-16 by thisiskeithb

marcusletric, 2020-10-16:
Alright then, thanks and sorry for the inconvenience.