MarlinFirmware/Marlin #19774: [BUG] M410 during home doesn't stop the whole move

schlaegerz, 2020-10-16:
All relevant config and stuff is from this one: https://github.com/MarlinFirmware/Marlin/issues/19490

### Bug Description
If you try to use emergency parser M410 it will only cancel the active move inside of a homing command

### My Configurations

https://github.com/MarlinFirmware/Marlin/files/5270184/Marlin.zip

### Steps to Reproduce

G28
wait a few seconds
M410

Printer should stop completely, but the printer will do the remaining moves it hasn't accomplished.

I assume this is a bigger issue than just home command, but home is the most obvious case.

ellensp, 2020-10-17:
config file indicate this is a old version of marlin. Many issues with M410 have been recently been addresses. Also emergency passer for STM32 is only recently implemented. 

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

schlaegerz, 2020-10-19:
Yep I have tested it on the most recent, this bug actually came out of this implementation 
https://github.com/MarlinFirmware/Marlin/pull/19752

rhapsodyv, 2020-10-19:
> Yep I have tested it on the most recent, this bug actually came out of this implementation
> #19752

This bug don't came out of that implementation. `M410` never worked with home (and other complex move operations). I just fixed the code to `M410` actual run, as it not even ran.

Stop saying and repeating that I broke `M410`. Stop being annoying.

schlaegerz, 2020-10-19:
My suggested fix would be something like this:
```
  #if ENABLED(EMERGENCY_PARSER)
    kill(M112_KILL_STR, nullptr, true);	    if (emergency_parser.killed_by_M112) kill(M112_KILL_STR, nullptr, true);

    if (emergency_parser.quickstop_by_M410 && !emergency_parser.mid_stop) {
      emergency_parser.mid_stop = true; // quickstop_stepper may call idle so make sure we don't go in an infinite loop
      quickstop_stepper();
       emergency_parser.mid_stop = false; 
    }
  #endif
```
// End up gcode main functon
#if ENABLED(EMERGENCY_PARSER)
emergency_parser.quickstop_by_M410 = false;
#endif
I don't know much about Marlin code standards and stuff and don't quite have the time to push through a PR, but maybe someone could use this as a jumping off point.

closed 2020-11-26 by github-actions[bot]