MarlinFirmware/Marlin #19775: [BUG] axis_codes[] out of bounds when outputting stepper motor currents with M503

nuevocampo, 2020-10-16:
### Bug Description

M503 produces garbled output. This is caused by:

`settings.cpp > void MarlinSettings::report()`
[...]
`3731: SERIAL_CHAR(axis_codes[q]);`

where the loop is enumerated out the bounds of `axis_codes[4]` when`MOTOR_CURRENT_COUNT > 4`

### Steps to Reproduce

1. Navigate to OctoPrint terminal (haven't tried other terminals)
2. Enter M503

**Expected behavior:** [What you expect to happen]

Output:
Recv: echo:; Stepper motor currents:
Recv: echo:  M907 X135 Y135 Z135 E135 E135

**Actual behavior:** [What actually happens]

Output:
Recv: echo:; Stepper motor currents:
M907 X135 Y135 Z135 E135 \x87135ok

after which the terminal stops responding.

rhapsodyv, 2020-10-16:
Please attach your configs.

nuevocampo, 2020-10-16:
Configs for this are irrelevant. It's a simple bug in code, immediately obvious when you look at the implementation:

```
LOOP_L_N(q, MOTOR_CURRENT_COUNT) {
    SERIAL_CHAR(' ');
    SERIAL_CHAR(axis_codes[q]); // <--- here, if q >= length(axis_codes)
    SERIAL_ECHO(stepper.motor_current_setting[q]);
}

```

rhapsodyv, 2020-10-16:
I didn't look at the code, but maybe `MOTOR_CURRENT_COUNT` >= `length(axis_codes)` is invalid. 
And we can't not randomly search for code that match your description.

If you have a valid use case, send your configs, please, and follow the issue template.

boelle, 2020-10-17:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Without these we CAN'T help you, they are also required by the issue template that you read ??

thisiskeithb, 2020-10-17:
If you can't be bothered to fill out the template & attach your configs, we can't be bothered to help.

closed 2020-10-17 by thisiskeithb

nuevocampo, 2020-10-17:
This was introduced with commit 492ba2a11 by @thinkyhead

sjasonsmith, 2020-10-17:
@nuevocampo please just attach your configs and we can reopen this. It seems like something that should be fixed.

Configs might seem irrelevant, but they never are. Even just changing the motherboard name is enough to justify attaching a config.

Since you seem to already know how to solve the problem, it would be best for you to just open a pull request with the change. Of course that template will want your configs as well.

nuevocampo, 2020-10-17:
Sure, I get it, thanks for the followup. I'm in middle of modifying the configs for my printer so they're in unclean state atm :)