MarlinFirmware/Marlin #19780: [BUG] MKS Robin Nano 1.2 + Color UI - no remaining time

bkiziuk, 2020-10-17:
### Bug Description

For configuration:
```
#define MKS_ROBIN_TFT35
#define TFT_COLOR_UI
```
the option `SHOW_REMAINING_TIME` is not active due to
`#if EITHER(SDSUPPORT, LCD_SET_PROGRESS_MANUALLY) && ANY(HAS_MARLINUI_U8GLIB, HAS_MARLINUI_HD44780, IS_TFTGLCD_PANEL)` seem to return FALSE.
Therefore, the UI shows only elapsed time.

### Steps to Reproduce

1. Generate G-Code with M73 option enabled
2. Start printing
3. Observe print progress on display

**Expected behavior:** 

Display shows both Elapsed time and Remaining time.

**Actual behavior:** 

Display shows only a progress bar and elapsed time.
[bugfix_2-0-x_20201015.zip](https://github.com/MarlinFirmware/Marlin/files/5395525/bugfix_2-0-x_20201015.zip)

rhapsodyv, 2020-10-17:
Is it a feature request or a bug? 🤔

bkiziuk, 2020-10-17:
Taking into account that some LCD definitions have been added/changed in the recent bugfix-2.0.x and the preprocessor's directives seem to not fully follow it - I would consider it as a bug. 
But considering the fact that the Touch UI is new, maybe it just does not support it yet.

rhapsodyv, 2020-10-17:
I did some search, and the Color Ui don't have the code to handle SHOW_REMANING_TIME. So, the checks are right: it isn't supported yet.

Please change to feature request. 

Anyway, I will try to work on that soon.

Thanks!

dchauran, 2021-02-27:
@rhapsodyv are you still planning to work on this soon? If not, should I take a stab at it?

thisiskeithb, 2022-10-08:
See https://github.com/MarlinFirmware/Marlin/pull/24843

PR #24843 (2023-06-23): Color UI remaining time
### Description
Add the possibility of displaying remaining print time with `TFT_COLOR_UI`.

The time remaining logic was copied from DOGM code, as it seemed the most complete - either taking time from M73 or extrapolating time elapsed by print progress.

Additionally the display items (in particular the feed/flow rate) are slightly rearranged to make room for elapsed/remaining time icons.

### Requirements
Requires an LCD display.

### Benefits
Fixes #19780, improves the display usability/readability.

### Configurations
I'm testing this with MKS Robin Nano 1.2 and its [... cut]
closed 2022-10-08 by thisiskeithb