MarlinFirmware/Marlin #19783: Strange temperature limit @284C

Drzet, 2020-10-17:
### Bug Description
Newly compiled Marlin 2.0.x bugfix.

Despite #define HEATER_0_MAXTEMP set to 350 whenever I issue M104 or M109 command with any value above 284, it gets reduced to 284C. I use PT1000 thermistor and full metal hotend and used to print polycarbonate a lot.
Really strange

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5396139/config.zip)

### Steps to Reproduce

Send: M104 S300 T0
Recv: ok
Recv:  T:22.28 /284.00

**Expected behavior:** [What you expect to happen]
Heat up to 300C
**Actual behavior:** [What actually happens]
Temp is limited to 284C. No errors displayed

swilkens, 2020-10-18:
I believe we set a protective limit ~15 degrees below the defined HEATER_0_MAXTEMP

See, for example: https://github.com/MarlinFirmware/Marlin/issues/17693

ellensp, 2020-10-18:
Two things are going one.

1) as above, your limited to -15 from max temp, to allow for overshoot etc
2) the max temp your setting is being overridden to the max temp in thermistor table 1010. which is 300c

closed 2020-10-18 by ellensp

Drzet, 2020-10-18:
Thank you for the explanation. Is there any particular reason why 1010 table stops at 300°C while 1047 table for the same thermistor goes all the way to 500?