MarlinFirmware/Marlin #19785: [FR] (Tramming Assistant Auto-Correct Points)

yysh12, 2020-10-17:
### Description
in g25.cpp line 130:
130: const float z_probed_height = probe.probe_at_point(screws_tilt_adjust_pos[i], PROBE_PT_RAISE, 0, true);

it should first check that the point provided by screws_tilt_adjust_pos[i], which uses the hotend as reference, is accessible by probe by taking into account the bed size and the probe x and y offsets. If it is not, it should truncate/modify the value automatically to the nearest probe accessible point. 

For example, if 235 is max X position by hot end, but Z x offset is -43, then the maximum accessible x position by the probe is 192. If a user sets the x point to 195 then G35 will fail with no explanation that the probe offset was the problem. You're average user could be confused by this. 

Instead it should give a warning stating that supplied point is inaccessible and that the nearest possible point will be utilized instead. Using the nearest possible point makes more sense than failing because at least some value can extracted from tramming assistant. And if they are like me, my point was just barely out of reach and using the nearest point made no functional difference on the utility of the tramming assistant. But even if far out, a visual warning in addition to nearest point still makes more logical sense. 

### Feature Workflow

1. Check if screws_tilt_adjust_pos[i] is an accessible position
2. IF YES: continue as currently programmed to do
3. IF NO: 
4. Alert user point is inaccessible given current probe offset settings and that the nearest value will be utilized instead
5. Automatically adjust screws_tilt_adjust_pos[i] to the nearest probe accessible point

#### Additional Information
* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Speaka, 2020-10-29:
Can't you just check if `#define TRAMMING_POINT_XY{...}` are inside `#define X_BED_SIZE 200
#define Y_BED_SIZE 200`  ? I dont know if it's done in sanitychecks but his seems sufficient for me. If this is not done maybe this should be?

Maybe there's a problem:
Ideally the tramming points are exactly above the bed screws to calculate the really needed screw turns. Some bedlayouts dont allow to print above the screws, so the tramming points may be outside the print area.
How would you define a limit if the probe location is valid outside the printarea?

So if you set the tramming points inside the bedarea you will have a reachable probing point. Youl get a little offset in adjustment turns but this seems to be negligible to me.

thisiskeithb, 2020-10-29:
> Can't you just check if #define TRAMMING_POINT_XY{...} are inside #define X_BED_SIZE 200 #define Y_BED_SIZE 200

There's no compile-time checks and runtime probe offset changes make this difficult to check for as well.

> So if you set the tramming points inside the bedarea you will have a reachable probing point. 

Not necessarily. If your probe is to the left of your nozzle, your nozzle/carriage may not be able to reach the tramming points on the right side of the bed due to `X_MAX_POS` and same goes if your probe is positioned on the right side, it could be limited by your `X_MIN_POS` and won't be able to reach the points on the left.

Speaka, 2020-10-29:
Probing returns NAN 
if the poitn is not reachable [here](https://github.com/MarlinFirmware/Marlin/blob/1247f7375402809034843088617d30fb79985cee/Marlin/src/module/probe.cpp#L671-L678)
This is checked in G35. Which reports just that it fails nto why. 
Besides this, NAN is returned if an error occures when deploy/stow fails. So we need to drill up the  `probe_at_point` function to return more precise errors. How should this be done correctly?

Speaka, 2020-10-30:
If you are in debug it wil [tell you](https://github.com/MarlinFirmware/Marlin/blob/6071a0835ad063ae7bdf1853ba2ca27b78493f64/Marlin/src/module/probe.cpp#L674) why its failing. 
I don't know if it's necessary to lift this output from debug to production. What do you think?
Positioning your Bed, and get the probe offset right is a workflow which isn't usually done in "non debug" since it doesnt change if its once setup. 
Fallback to the next reachable position will in the worst case fail the calculations since the screw might be a few cm away.

Also the initial problem should occur on every call of `probe_at_point` if I get it right. E.g. G29, G30, M48 and so on.

Lmk if you have other opinion.

yysh12, 2020-10-30:
No it should be necessary. Average user won't know what is happening. I understand this feature is still in its infancy, but still some dialog should be returned. Also, I still think autocorrecting points with a warning is still better than nothing. The fact the feature is enabled should mean that some value should be extracted from it. It is then up to user to modify hardware to ensure they can probe the requisite points. But the software should be a bit more accommodating. Just my opinion.

qwewer0, 2020-10-30:
One thing that I could think that would be wrong with autocorrecting points, is that If the autocorrected points (two or more) aren't directly above the screws then the calculated turns will be incorrect.
If all the points would be modified by the same amount from the screws, than the turns would be equally wrong, which could be useable, but still not ideal, on the other hand if only two or three points would be changed, than the turn values would also be wrong in general and even to other screws.

Speaka, 2020-10-30:
I agree with you that the error should be more meaningful. Maybe changing the output from debugging to regular reporting will hint towards the right information. "Position ot reachable." Other probing error scenarios should be coverd then also with this.
If you know you cant reach the position you can adjust the probing points.

My suggestion: Report that the point cant be rached and what distance is missing. If the desired probing point for example is outside the bed, it will report something like _"Error, Position not reachable. 2mm outside Ymin"_

yysh12, 2020-11-02:
Yeah, regardless of autocorrect points, that would still be very useful to get people directed the right way.

closed 2020-12-24 by yysh12